raspberrypi/linux #1607: WARNING: CPU: 1 PID: 1167 at net/wireless/sme.c:981 cfg80211_connect+0x234/0x588 [cfg80211]()

vorburger, 2016-08-20:
https://bugzilla.kernel.org/show_bug.cgi?id=153471 : I'm trying to use a Raspberry Pi as a WiFi Wireless Access Point AP with hostapd and, like another user on https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=143972&p=1021341, see wlan0 going down with the following shown in dmesg:

[Sat Aug 20 18:15:08 2016] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[Sat Aug 20 18:15:08 2016] brcmfmac: brcmf_add_if: ignore IF event
[Sat Aug 20 18:15:09 2016] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[Sat Aug 20 18:15:10 2016] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Aug 20 18:15:10 2016] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Aug 20 18:15:10 2016] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Aug 20 18:15:11 2016] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Aug 20 18:15:12 2016] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x41E1
[Sat Aug 20 18:15:18 2016] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[Sat Aug 20 18:15:18 2016] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Aug 20 18:15:19 2016] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Sat Aug 20 18:15:19 2016] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[Sat Aug 20 18:15:20 2016] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x41E1
[Sat Aug 20 18:15:22 2016] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
[Sat Aug 20 18:15:22 2016] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
[Sat Aug 20 18:15:22 2016] ------------[ cut here ]------------
[Sat Aug 20 18:15:22 2016] WARNING: CPU: 1 PID: 1167 at net/wireless/sme.c:981 cfg80211_connect+0x234/0x588 [cfg80211]()
[Sat Aug 20 18:15:22 2016] Modules linked in: bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 rfkill evdev snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio ipv6
[Sat Aug 20 18:15:22 2016] CPU: 1 PID: 1167 Comm: iwconfig Not tainted 4.4.18-v7+ #905
[Sat Aug 20 18:15:22 2016] Hardware name: BCM2709
[Sat Aug 20 18:15:22 2016] [<80018784>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
[Sat Aug 20 18:15:22 2016] [<80014058>] (show_stack) from [<80320bc4>] (dump_stack+0xd4/0x118)
[Sat Aug 20 18:15:22 2016] [<80320bc4>] (dump_stack) from [<80025360>] (warn_slowpath_common+0x98/0xc8)
[Sat Aug 20 18:15:22 2016] [<80025360>] (warn_slowpath_common) from [<8002544c>] (warn_slowpath_null+0x2c/0x34)
[Sat Aug 20 18:15:22 2016] [<8002544c>] (warn_slowpath_null) from [<7f11cad4>] (cfg80211_connect+0x234/0x588 [cfg80211])
[Sat Aug 20 18:15:22 2016] [<7f11cad4>] (cfg80211_connect [cfg80211]) from [<7f135c5c>] (cfg80211_mgd_wext_connect+0x114/0x180 [cfg80211])
[Sat Aug 20 18:15:22 2016] [<7f135c5c>] (cfg80211_mgd_wext_connect [cfg80211]) from [<7f135d8c>] (cfg80211_mgd_wext_siwfreq+0xc4/0x194 [cfg80211])
[Sat Aug 20 18:15:22 2016] [<7f135d8c>] (cfg80211_mgd_wext_siwfreq [cfg80211]) from [<7f135adc>] (cfg80211_wext_siwfreq+0xc8/0x134 [cfg80211])
[Sat Aug 20 18:15:22 2016] [<7f135adc>] (cfg80211_wext_siwfreq [cfg80211]) from [<805ae300>] (ioctl_standard_call+0x6c/0x4b4)
[Sat Aug 20 18:15:22 2016] [<805ae300>] (ioctl_standard_call) from [<805aea40>] (wext_handle_ioctl+0x1b8/0x23c)
[Sat Aug 20 18:15:22 2016] [<805aea40>] (wext_handle_ioctl) from [<804ee504>] (dev_ioctl+0x53c/0x800)
[Sat Aug 20 18:15:22 2016] [<804ee504>] (dev_ioctl) from [<804b7638>] (sock_ioctl+0x12c/0x2b0)
[Sat Aug 20 18:15:22 2016] [<804b7638>] (sock_ioctl) from [<80169720>] (do_vfs_ioctl+0x424/0x614)
[Sat Aug 20 18:15:22 2016] [<80169720>] (do_vfs_ioctl) from [<80169954>] (SyS_ioctl+0x44/0x6c)
[Sat Aug 20 18:15:22 2016] [<80169954>] (SyS_ioctl) from [<8000fb40>] (ret_fast_syscall+0x0/0x1c)
[Sat Aug 20 18:15:22 2016] ---[ end trace dac61f84813e0313 ]---
[Sat Aug 20 18:15:22 2016] ------------[ cut here ]------------

vorburger, 2016-08-21:
https://bugzilla.kernel.org/show_bug.cgi?id=153471 recommends to first try a newer Kernel..

Ruffio, 2017-02-05:
@vorburger Can this issue be closed?

vorburger, 2017-02-07:
@Ruffio yeah, closed it; this was due to bad hardware ...

closed 2017-02-07 by vorburger