raspberrypi/linux #1609: Modulize Goodix Touchscreen driver not working with Rpi

explorerE, 2016-08-22:
Goodix.dts configuration file not fount overlay folder

pelwell, 2016-08-22:
Is there any reason it should be there? If this is a request for an overlay or a driver to be added to the kernel then you will have to explain what the device is and why we should support it. If you don't have an overlay to add then you might be able to get some help in the Raspberry Pi Forums - the [Advanced Users](https://www.raspberrypi.org/forums/viewforum.php?f=29) and [Interfacing](https://www.raspberrypi.org/forums/viewforum.php?f=44) sections are good places to start (but choose one - don't post the same topic to both).

Ruffio, 2016-08-29:
@explorerE can this issue be closed? If yes, then please do so. Thanks.

xdsopl, 2017-05-15:
I have a working goodix.dts file but it is not generic and works only for our Artista-IoT.
I was thinking to make it generic using the default pins:
1,3,5,7,9,11 (+3V3, GPIO2/SDA1, GPIO3/SCL1, GPIO4/INT, GND, GPIO17/RST)
At the moment it is a PITA to get a Linux kernel with Goodix module on Raspbian but that's the way it is for now.
What would be necessary to get the Goodix module included in Raspbian?
I did talk to my superiors but we are B2B so it wont be so easy to get a touch controller as a non business person and therefore of limited profit to the community.
There are plenty of Goodix controllers you could salvage from broken touch devices though.
Ahmet

popcornmix, 2017-05-15:
> At the moment it is a PITA to get a Linux kernel with Goodix module on Raspbian but that's the way it is for now.
What would be necessary to get the Goodix module included in Raspbian?

Is enabling this config option enough? ``CONFIG_TOUCHSCREEN_GOODIX``

xdsopl, 2017-05-15:
Yes, that would be awesome.
Should I put my working dts file here as reference?
Ahmet

pelwell, 2017-05-15:
Please create a pull request if you have a working overlay (not a modified base dts).

xdsopl, 2017-05-15:
Thanks, will do.
This way we could even get the useful back-ports for the Goodix module into 4.4?

Ahmet

pelwell, 2017-05-15:
I don't want to commit to that. Why don't you want 4.9?

xdsopl, 2017-05-15:
Personally I want 4.9 to be default ASAP. As long as it is not default ..
But you're right, switching to 4.9 is easy using raspi-config.
Will come back to you with a pull request.
Thank you.
Ahmet

pelwell, 2017-05-15:
FYI, `apt-get upgrade` now also installs a 4.9 kernel. Only the images have yet to be updated, and that won't be true for very long.

xdsopl, 2017-05-15:
Awesome :)
Ahmet

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65