raspberrypi/linux #1610: ARM64 Kernel Support

Electron752, 2016-08-22:
I have some changes to the kernel to enable arm64 support in many of the downstream drivers.   This is complementary to the upstream work in 4.8 but involves drivers that can't or won't be upstreamed at this time.

1. Any interest in adding changes to the downstream tree to support arm64?
2. Would it be better to submit this as a single change or a few small changes?
3. Should I just submit a pull request?

Here are the drivers that I've tested to be working.
1. BCM2709 DMA API - Needed to make the FB video driver work correctly without any changes.
2. DWC_OTG USB
3. FT5406 touchscreen support
4. Limited VCHIQ support.
5. HDMI audio.

These drivers just work without any needed changes
1. bcm2835-mmc
2. bcm2835-sdhost
3. Builtin bluetooth
4. Builtin wifi
5. VC4 DRM driver for advanced video and opengl support.

popcornmix, 2016-08-22:
I think submitting a PR may be best. Without seeing the scope of the changes it is hard to say whether they would be accepted.
In general if there are small patches to a driver with no change in behaviour for 32-bit users, then I don't see a problem.
Obviously we rebase on upstream code periodically so changes to upstream code should ideally come from upstream (or be submitted upstream too).
Drivers that are only downstream can be changed more easily.
Start with a single PR. If some parts are more controversial that others the PR can be split to get the easier bits merged, but that may not be necessary.

Electron752, 2016-08-22:
I submitted a pull request, but the git hub interface makes this all look a lot worse then it actually is.   If you clone to the local disk and use gitk or meld, everything makes much more sense.

Also, the sheer number of lines makes it worse as well.   I think everything would be much clearer to do it in pieces. I would rather do it that way then have everyone be shocked and discouraged.

Also, if some of the changes could be propagated to 32 bit then it make a huge difference in the number of required #ifdefs.  For example, on 64 bit it's no longer permitted to pass a NULL context to the dma buffer allocation functions.  So the workaround is to carry some extra state around which makes things a bit messy.

clivem, 2016-08-22:
Hmmm. I wonder how not having FIQ is going to effect USB audio? I know you said in the PR, that "Even though FIQ is not used on 64bit, the driver still outperforms the upstream driver", but I wonder whether anyone has actually tested USB audio without FIQ? ;)

Electron752, 2016-08-22:
I can certainly test that...   It may also be interesting to take the dwc_otg driver out of the mix for now and focus on the other parts.  People can always use the upstream driver which isn't great but works ok.

BTW, I think USB audio is foobar even with FIQ.   I've seen lots and lots of issues with the dwc_otg driver where it doesn't schedule around SOF very well.  Since most USB audio is full speed rather then high speed, I could certainly see issues.

clivem, 2016-08-22:
> BTW, I think USB audio is foobar even with FIQ.

LOL. I'd agree with that. However, I just remember how much more of a train-wreck it was, prior to the implementation of FIQ. ;)

Electron752, 2016-08-22:
I guess that's why these things still have analog audio jacks...

I don't think USB audio is that popular with the general crowd.  Some people are all for it.

popcornmix, 2016-08-22:
To be honest in the Pi1 days the most obvious issue with the usb driver was keyboards. If you lose a key up message you type in ``ls`` and get ``lssssssssssssssssssssssssssssssssssssssssssssssssss``.

And yes without a FIQ I suspect that will still happen when system is under load.

Electron752, 2016-08-22:
LOL. The keyboard thing is funny.

Like I said, I can do more investigation into FIQ on arm64, but I think it was dropped from the hardware.  At least I couldn't find any documentation on it.  It may only be available in a complete 32 bit environment.

Electron752, 2016-08-22:
BTW, I don't know if you follow the forum, but Microchip makes smarter USB hubs now that would also reduce the need for FIQ.

popcornmix, 2016-08-22:
Pretty sure FIQ is supported by the hardware when arm is in 64-bit mode.
I think the problem is just that upstream kernel hasn't supported it so far.

Electron752, 2016-08-22:
I propose then that we drop the USB support for now and just use the downstream driver until I can do more research.   Doing a full FIQ port would also minimize the number of needed changes.

I've never done a pull request on github before.  Do I incorporate all the comments and resubmit a new pull request or do I somehow revise this pull request?

popcornmix, 2016-08-22:
In general you can edit you branch (``Electron752:rpi-4.6.y+rpi364-rebase6``), e.g. with ``git rebase -i HEAD~`` to make it look how you want, and then ``git push -f`` and this PR will update.
Feel free to do that as often as you need. You never need to create a new PR for the same purpose.

Ideally you should split the commits into logical chunks (e.g. add arm64 config files. Add arm64 device tree. arm64 support for dma driver. arm64 support for sound driver etc.  

However it could be worthwhile here to create smaller PRs with some of these commits that are easier to merge. When they are merged you can rebase your tree and the main PR will get smaller.

Electron752, 2016-08-22:
OK cool.  

Let's table the review at this point while I make some changes based on the feedback.   I'm planning to go offline a bit anyway.

Thanks for all your feedback!

closed 2016-08-23 by Electron752

SuperWangCC, 2016-08-23:
Yes I think 
###so.ARM64 device not only need kernel support but need system support so.
But many kernel libs did not port to 64bits.This need Raspberry pi Foundation 's support.Such as Gcc .
I remember they said that will support arm64.
Proformance up,We need 64bits.

pelwell, 2016-08-23:
No, the Foundation have not committed to supporting 64-bit kernels. If a clear performance benefit can be demonstrated and we can continue to support all our existing devices and use cases then we would consider it, but at the moment that is not the case.

Ferroin, 2016-08-23:
That then begs the question of what qualifies as a clear performance benefit.  On x86 there's a significant measurable performance boost running 64-bit, but that's mostly because 64-bit x86 has 8 more general purpose registers to work with.  Such significant architectural differences don't exist in ARMv8 between 32 and 64 bit modes, so your not as likely to see such an improvement.  Areas that should see measurable performance improvement on ARM are mostly syscalls, specifically ones that deal with file offsets and time-stamps.  stat(2) in particular is faster on 64-bit, regardless of the underlying architecture.

The other big area of improvement though is something most people don't think about in this context: security.  64-bits means you can address more RAM, but it also means you have a larger virtual address space, which means ASLR is a better defense (would you rather hunt through 3GB for an exploit, or 128TB?).

pelwell, 2016-08-23:
It's always going to be somewhat subjective, but we'd want to see a series of varied benchmarks, some of which are of real-world use cases, with hard numbers before we commit. We would probably tolerate a few small performance regressions, but feature regressions would be hard to justify.

pelwell, 2016-08-23:
For example, lack of FIQ support would probably be a deal-breaker.

grigorig, 2016-08-25:
What about using a 64-bit kernel with 32 bit userspace? I remember on x86 that used to have some advantages. Is the same true for AArch64?

popcornmix, 2016-08-25:
32-bit userspace has been reported as working with a 64-bit kernel.
But still will have the issue of the non-FIQ USB and no GPU accelerated video and camera.

The advantage on x86 would be supporting more than 4G of RAM (although less than 4G in a single process), but that won't help on Pi.

pelwell, 2016-08-25:
Even on a Pi2 we found the 3G/1G User/Kernel split too limiting and chose to switch to 2G/2G to avoid using HIGHMEM. We wouldn't have that problem with a 64-bit kernel.

pelwell, 2016-08-25:
FYI @grigorig the (currently unsupported) ARM64 kernel configuration merged yesterday (https://github.com/raspberrypi/linux/blob/rpi-4.8.y/arch/arm64/configs/bcmrpi3_defconfig) defines CONFIG_COMPAT, which enables support for 32-bit userspace.

ericcurtin, 2016-11-28:
Does anyone know where this semi-working 64 bit kernel went to that I keep hearing about?  Electron752 @pelwell @popcornmix @grigorig @Ferroin @SuperWangCC @clivem ?

pelwell, 2016-11-29:
You can build it from the rpi-4.8.y branch. This works for me:
```
make ARCH=arm64 CROSS_COMPILE=~/x-tools/aarch64-unknown-linux-gnueabi/bin/aarch64-unknown-linux-gnueabi- bcmrpi3_defconfig Image
cp arch/arm64/boot/Image kernel8.img
```

ericcurtin, 2016-11-29:
Cool, thanks @pelwell this should prove very useful. We've been hoping we could use the RPI3 as a test device for a more expensive piece of aarch64 hardware. Now we have a kernel great!

ericcurtin, 2016-11-29:
I'm using Ubuntu /usr/bin/aarch64-linux-gnu-gcc binary from 16.04 repo. Hopefully this works. End goal is trying to get Ubuntu 14.04 aarch64 OS running on RPI 3. Might have to over come a few hurdles first. Kernel seems to be building fine anyway!