raspberrypi/linux #1616: Dealing with the diffent way the kernel build system handles dsb between arm and arm64?

Electron752, 2016-08-23:
One of the major portability issues I'm noticing is the way that the kernel build headers do not support dsb without an option on arm64 anymore.

I read and think I understand the aarch64 reference manual and I think it says that dsb without an option should be understood as dsb(sy).   So I'm wondering if it would be better to change all the dsb() to dsb(sy) in the downstream code or add #ifdef CONFIG_ARM64?

I noticed in the upstream code that these are being converted to other macros such as wmb or rmb.  I'm wondering if the point of removing support for dsb without an option was an attempt to force people to convert to more portable macros?

pelwell, 2016-08-23:
I also think that dsb should be replaced by dsb(sy). If that causes no instruction changes on ARM32 then I'm happy for use to make that switch now.

If there is a more portable version of dsb/dsb(sy), ideally one that generates the same instructions, then lets switch to that instead. From what I can tell, on ARMv7 the mapping is:
```
dsb   dsb(sy)
mb    dsb(sy)
rmb   dsb(sy)
wmb   dsb(st)
```
I'm surprised at rmb generating dsb and not dmb, but since dsb is stronger (more restrictive) than dmb it must have been chosen for a reason.

wmb is a weaker barrier than dsb/dsb(sy), but mb or rmb look like suitable alternatives.

Electron752, 2016-08-23:
My personal opinion would be to just do a bulk search and replace and change all the  dsb to dsb(sy).  I suspect it's going to be awhile before these drivers get ported to a non-arm architecture.

I didn't do that in the most recent PR because I wasn't 100% sure it would be exactly the same op-codes in the binary.  I noticed the reference manual gives a few alternative encodings for dsb(sy), and I didn't want to take the time to compile the kernel both ways to prove that it's exactly the same.

popcornmix, 2016-08-23:
Currently:
```
00000170 <bcm2836_arm_irqchip_send_ipi>:
 170:   e1a0c00d        mov     ip, sp
 174:   e92dd8f0        push    {r4, r5, r6, r7, fp, ip, lr, pc}
 178:   e24cb004        sub     fp, ip, #4
 17c:   e52de004        push    {lr}            ; (str lr, [sp, #-4]!)
 180:   ebfffffe        bl      0 <__gnu_mcount_nc>
 184:   e3003000        movw    r3, #0
 188:   e3403000        movt    r3, #0
 18c:   e1a06000        mov     r6, r0
 190:   e5937004        ldr     r7, [r3, #4]
 194:   f57ff04f        dsb     sy
 198:   e3a03001        mov     r3, #1
 19c:   e3004000        movw    r4, #0
 1a0:   e1a05113        lsl     r5, r3, r1
 1a4:   e3404000        movt    r4, #0
 1a8:   e3e00000        mvn     r0, #0
 1ac:   ea000002        b       1bc <bcm2836_arm_irqchip_send_ipi+0x4c>
 1b0:   f57ff04e        dsb     st
 1b4:   e0873200        add     r3, r7, r0, lsl #4
 1b8:   e5835080        str     r5, [r3, #128]  ; 0x80
 1bc:   e2802001        add     r2, r0, #1
 1c0:   e3a01004        mov     r1, #4
 1c4:   e1a00006        mov     r0, r6
 1c8:   ebfffffe        bl      0 <_find_next_bit_le>
 1cc:   e5943000        ldr     r3, [r4]
 1d0:   e1530000        cmp     r3, r0
 1d4:   cafffff5        bgt     1b0 <bcm2836_arm_irqchip_send_ipi+0x40>
 1d8:   e89da8f0        ldm     sp, {r4, r5, r6, r7, fp, sp, pc}
```
and with ``dsb`` changed to ``dsb(sy)``
```
00000170 <bcm2836_arm_irqchip_send_ipi>:
 170:   e1a0c00d        mov     ip, sp
 174:   e92dd8f0        push    {r4, r5, r6, r7, fp, ip, lr, pc}
 178:   e24cb004        sub     fp, ip, #4
 17c:   e52de004        push    {lr}            ; (str lr, [sp, #-4]!)
 180:   ebfffffe        bl      0 <__gnu_mcount_nc>
 184:   e3003000        movw    r3, #0
 188:   e3403000        movt    r3, #0
 18c:   e1a06000        mov     r6, r0
 190:   e5937004        ldr     r7, [r3, #4]
 194:   f57ff04f        dsb     sy
 198:   e3a03001        mov     r3, #1
 19c:   e3004000        movw    r4, #0
 1a0:   e1a05113        lsl     r5, r3, r1
 1a4:   e3404000        movt    r4, #0
 1a8:   e3e00000        mvn     r0, #0
 1ac:   ea000002        b       1bc <bcm2836_arm_irqchip_send_ipi+0x4c>
 1b0:   f57ff04e        dsb     st
 1b4:   e0873200        add     r3, r7, r0, lsl #4
 1b8:   e5835080        str     r5, [r3, #128]  ; 0x80
 1bc:   e2802001        add     r2, r0, #1
 1c0:   e3a01004        mov     r1, #4
 1c4:   e1a00006        mov     r0, r6
 1c8:   ebfffffe        bl      0 <_find_next_bit_le>
 1cc:   e5943000        ldr     r3, [r4]
 1d0:   e1530000        cmp     r3, r0
 1d4:   cafffff5        bgt     1b0 <bcm2836_arm_irqchip_send_ipi+0x40>
 1d8:   e89da8f0        ldm     sp, {r4, r5, r6, r7, fp, sp, pc}
```
i.e. it will be binary identical. So I'm okay with changing ``dsb`` to ``dsb(sy)`` in both the arm32 and arm64 cases.

popcornmix, 2016-08-23:
On Pi1:
With ``dsb()``
```
000000a4 <bcm_dma_wait_idle>:
  a4:   e1a0c00d        mov     ip, sp
  a8:   e92dd800        push    {fp, ip, lr, pc}
  ac:   e24cb004        sub     fp, ip, #4
  b0:   e52de004        push    {lr}            ; (str lr, [sp, #-4]!)
  b4:   ebfffffe        bl      0 <__gnu_mcount_nc>
  b8:   e3a02000        mov     r2, #0
  bc:   ee072f9a        mcr     15, 0, r2, cr7, cr10, {4}
  c0:   e5903000        ldr     r3, [r0]
  c4:   ee072f9a        mcr     15, 0, r2, cr7, cr10, {4}
  c8:   e3130001        tst     r3, #1
  cc:   089da800        ldmeq   sp, {fp, sp, pc}
  d0:   e5903000        ldr     r3, [r0]
  d4:   ee072f9a        mcr     15, 0, r2, cr7, cr10, {4}
  d8:   e3130001        tst     r3, #1
  dc:   1afffffb        bne     d0 <bcm_dma_wait_idle+0x2c>
  e0:   e89da800        ldm     sp, {fp, sp, pc}
```
With ``dsb(sy)``
```
000000a4 <bcm_dma_wait_idle>:
  a4:   e1a0c00d        mov     ip, sp
  a8:   e92dd800        push    {fp, ip, lr, pc}
  ac:   e24cb004        sub     fp, ip, #4
  b0:   e52de004        push    {lr}            ; (str lr, [sp, #-4]!)
  b4:   ebfffffe        bl      0 <__gnu_mcount_nc>
  b8:   e3a02000        mov     r2, #0
  bc:   ee072f9a        mcr     15, 0, r2, cr7, cr10, {4}
  c0:   e5903000        ldr     r3, [r0]
  c4:   ee072f9a        mcr     15, 0, r2, cr7, cr10, {4}
  c8:   e3130001        tst     r3, #1
  cc:   089da800        ldmeq   sp, {fp, sp, pc}
  d0:   e5903000        ldr     r3, [r0]
  d4:   ee072f9a        mcr     15, 0, r2, cr7, cr10, {4}
  d8:   e3130001        tst     r3, #1
  dc:   1afffffb        bne     d0 <bcm_dma_wait_idle+0x2c>
  e0:   e89da800        ldm     sp, {fp, sp, pc}
```
Which looks okay too.

Electron752, 2016-08-23:
Do you want me to update the change again to remove the #ifdef CONFIG_ARM64?

popcornmix, 2016-08-23:
Yes. You can replace ``dsb`` with ``dsb(sy)`` without the ``#ifdef CONFIG_ARM64``

Electron752, 2016-08-23:
OK, I made the replacements and pushed a new forced update.  It simplifies the patches alot.

Electron752, 2016-08-23:
I also updated the pull request with the latest round of changes.

Electron752, 2016-08-23:
At this point I think my changes are very simple, make it through the formating/style tests, and should be very safe for not breaking anything 32 bit.

At this point I want to ask again if their is any interest in accepting changes for arm64 from the community?   I really don't want to waste my time and and the time of others if I'm barking up the wrong tree.

Thanks.

popcornmix, 2016-08-23:
I think having whatever patches are required to build a 64-bit kernel in the raspberrypi/linux tree makes sense if they don't cause problems. That allows users and other distributions to build 64-bit kernels without having to manage a separate set of patches that are likely to get outdated.

Whether raspbian switches to using a 64-bit by default is a separate question and isn't something that will happen soon. However having the patches required within the repo does remove one of the obstacles to making that decision.

Electron752, 2016-08-23:
I agree that having patches in the main tree benefits the community because it quickly becomes too much work to constantly fold in changes for something as dynamic as the linux kernel.

I also agree that converting raspbian to 64 bit doesn't make sense at this time since most users are not going to notice any improvement and will only notice that it breaks things.  But it's also in line with the goals of education because it gives people in the community something to tinker with.

Are there any other issues with the patches that I've submitted?  I have some patches for the device tree and a 64 bit config file that I have ready as well.  I was planing to submit it tomorrow.  That should make it possible for people to build a "working" 64 bit kernel based on the downstream tree.

popcornmix, 2016-08-23:
Yes, 64-bit config and device tree files should be fine for the next step.

closed 2016-08-24 by Electron752