raspberrypi/linux #1619: ARM vs ARM64 - Passing a NULL driver context to DMA functions

Electron752, 2016-08-24:
Hi,

I just wanted to bring up one more issue before I leave everyone alone for awhile.

A second major portability issue between arm32 and arm64 is passing a NULL driver to some of the various dma and memory allocation and deallocation functions.   On arm32 this works fine, but for reasons I don't totally understand is broken on arm64.

Most of the upstream kernel these days do pass a driver context for better reporting and debugging.  So I'm thinking this is being considered a "best practice" and being forced onto people.   

Unfortunately, this can be very messy to fix in a 64 bit specific way with #ifdef since it often requires carrying around more pointers.   A simple solution is to just store the pointer in a static global variable on 64 bit only since most of these drivers are have only have a single active instance.  But I'm not sure that is the best way to go.

pelwell, 2016-08-24:
Is there any harm in using a valid driver context on arm32? Can you highlight some examples of where it gets tricky?

Electron752, 2016-08-24:
No, no harm in passing it in for arm32.  I just think some of the downstream code was written when passing it in didn't have much advantage.

Here is an example from the MMC driver.   A function called log_init is called, but it doesn't have the driver context as a parameter so it passes NULL.   
`
static void log_init(u32 bus_to_phys)
{
        spin_lock_init(&log_lock);
        sdhost_log_buf = dma_zalloc_coherent(NULL, LOG_SIZE, &sdhost_log_addr,
                                             GFP_KERNEL);
        if (sdhost_log_buf) {
                pr_info("sdhost: log_buf @ %p (%x)\n",
                        sdhost_log_buf, sdhost_log_addr);
                timer_base = ioremap_nocache(bus_to_phys + 0x7e003000, SZ_4K);
                if (!timer_base)
                        pr_err("sdhost: failed to remap timer\n");
        }
        else
                pr_err("sdhost: failed to allocate log buf\n");
}
`
A better example would be dwc_otg.   Sometimes it does a dma memory allocation 10 levels or so deep on the stack and it would need to get that context passed in somehow.  In that driver I worked around it by stashing it in one of the structures, but the number of #ifdefs get messy.

pelwell, 2016-08-24:
That's my logging code - it did the job, but I'm happy to improve it.

Let's aim to also use valid device contexts on arm32, in the least hacky way that isn't painful.

Electron752, 2016-08-24:
OK, cool.

I'll be going offline for awhile...

I'm planning to get MMC/SD working again, then once that happens perhaps post something on the forum so that people that are interested and capable can try it out.

Of course, I'll post a disclaimer that everything is experimental and don't bother any of the main line developers with issues at this point.

closed 2016-08-24 by Electron752