raspberrypi/linux #1621: RTL8812 Support

arrmo, 2016-08-25:
Hi,

Struggling to configure the kernel to support the Realtek rtl8812au device (rtlwifi driver). More info here, but I can't seem to find this in the RPi kernel ... or perhaps I'm just missing it?
https://wiki.gentoo.org/wiki/AC1200_Wireless_Adapters

Thanks!

pelwell, 2016-08-25:
You could try using one of MrEngman's drivers - see this Forum thread: https://www.raspberrypi.org/forums/viewtopic.php?p=928025#p928025

arrmo, 2016-08-25:
Hi,

That makes sense, but isn't this support built in to the kernel now? I figured I'd just enable it and rebuild the kernel ... no?

Thanks!

pelwell, 2016-08-25:
> That makes sense, but isn't this support built in to the kernel now?

What makes you think that?

Diederik de Haas (@diederikdehaas) maintains a repository of with a driver for that device, with support up to Linux 4.7 - why would he do that if it was in the standard kernel? You can find his repo with instructions [here](https://github.com/diederikdehaas/rtl8812AU).

arrmo, 2016-08-25:
Just going by the info in the link I provided ... but I may be misunderstanding, definitely very possible! It just seemed like it was in the stock kernel now - no?

BTW, I can find many different drivers out there as well ... this one also seems very recent,
https://github.com/abperiasamy/rtl8812AU_8821AU_linux

Any thoughts on what the deltas are, and which one to use / trust? 

Thanks for the help!

clivem, 2016-08-25:
@arrmo To quote the Gentoo wiki page you referenced....

> Realtek 8723/8812/8821 PCIe drivers are in Linux 3.17.8 and later.
> The Realtek 8812au USB driver is available from https://github.com/gnab/rtl8812au

ie. their is an upstream driver for the PCI variant of the 8812 chipset, (CONFIG_RTL8821AE IIRC), but the 8812AU variant continues to be supported by an out-of-tree vendor driver and is probably best gotten from the link @pellwell referenced, as the guy maintaining this appears to have an interest in Debian/Raspbian.

arrmo, 2016-08-25:
Makes sense - thanks for all the help. Much appreciated!

closed 2016-08-25 by arrmo

pelwell, 2016-08-25:
I would consider MrEngman's and Diederik's drivers trustworthy - I can't speak for any others. MrEngman's are binary releases, which may be more convenient.

arrmo, 2016-08-25:
Appreciate the pointer - thanks!!!

diederikdehaas, 2016-08-25:
The driver repos from gnab and abperiasamy are based on a *much* older driver source then the one I'm maintaining.
As one of the developers of the [Raspbian NetInstaller](https://github.com/debian-pi/raspbian-ua-netinst) and general fan/supporter of Debian, Raspbian and Debian are indeed at the top of my prio list.

arrmo, 2016-08-25:
Awesome - thanks! Does your driver work fine with the latest RPi kernel (i.e. 4.4.19)?

diederikdehaas, 2016-08-25:
I have no reason to think it won't. 
If it doesn't then it's a bug and I'd welcome a bug report for it.

arrmo, 2016-08-25:
Works for me! I'll give it a go later today. Thanks for all the help!

arrmo, 2016-08-26:
It works - thanks! Am having a issue with DKMS build, will post that separately.