raspberrypi/linux #1622: Please support MT7610U Wifi USB dongle

adrianmay, 2016-08-27:
I'd like to make a wireless router based on a pi. If you shop on Amazon for a dongle with a powerful aerial, you'll probably get this chipset:

$ lsusb
Bus 001 Device 005: ID 0e8d:7610 MediaTek Inc.

Raspian supports MT7601, but that's completely different. I could probably add the driver source by following the pattern but I'm not sure how to get started cos I'm not seeing a source tree for my ubuntu box that spits out a SD image. If you don't have the dongle, I'll promptly test your builds with mine. Thanks!

popcornmix, 2016-08-27:
If the dongle is not supported by the upstream linux tree then we won't support it.
Either ensure your device is supported by upstream linux before buying it,
or request it that support is added to upstream linux and we will pick that up when moving to to a newer linux kernel.

pelwell, 2016-08-27:
I think MrEngman has a driver for this chip - see this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=107756

clivem, 2016-08-27:
@adrianmay I'm not sure whether you want to hear this, but for your usage, wireless router external antenna.....
What you really want is this.... 
http://www.amazon.co.uk/TP-LINK-TL-WN722N-150Mbps-Wireless-Adapter/dp/B002SZEOLG/
..... Yep, no 5G band support and 150Mbs rather than 300Mbs. But those "in the know", including the security pen test guys, and people who want bullet-proof wi-fi under Linux for semi-pro purposes, (which you do, if you are talking about making a wireless router with a Pi), use things which utilise the ath9k upstream driver. This is the last truly Open Source wifi hardware! Both the firmware and the driver are Open Source. The Alfa Network AWUS036NHA is the other "off-the-shelf" using the ath9k driver and having external antenna config, but at 20 quid it's twice as expensive as the £10TP-LINK. https://www.amazon.co.uk/dp/B004Y6MIXS/

@adrianmay  I know this advice is probably too late for you as you have already purchased a dongle, but it may help others doing what you want to do. If you want bullet-proof wifi under Linux, ath9k driver is the way to go for USB dongles and Intel when you can use mini-pcie hardware, but of course you don't have that option with the Pi. It is best to stay away from all those dongles requiring "out of tree" 3rd party vendor drivers, IMHO.

adrianmay, 2016-08-27:
> B002SZEOLG/

Hi Clive,

Thank you. That's exactly what I wanted to hear. I just ordered what you
suggested and I'll get a refund the other. These things are only a tenner a
pop anyway.

Adrian.

Ruffio, 2016-08-29:
@adrianmay so this issue can be closed now? If so, please do so. Thanks.

closed 2016-08-29 by adrianmay