raspberrypi/linux #1623: Glitches in HDMI audio output when in stress

softwarefailure, 2016-08-29:
I think there is an issue with the HDMI audio driver. I've created a small ALSA demo program which triggers the error. It's available here: http://www.softwarefailure.de/tmp/pisound.7z Please run it like this:

> ./alsastress bg.wav fx.wav
> or
> ./alsastress bg.wav fx2.wav

And then listen very carefully. Here there are clear audio glitches but only when using HDMI audio. There are no glitches when using analogue audio. 

Note that my demo program is extremely sloppy and there is no clean exit, just hit CTRL-C. Its only purpose is to demonstrate that the HDMI audio driver produces glitchy audio under stress. So please don't complain about the style :)

Also note that the glitches can only be heard if you listen carefully. I can hear the following glitches: The background music sometimes interrupts for a very short time and the fx samples sometimes don't play correctly. It sounds like something is cut off sometimes. But you need to listen carefully in order to notice it. And don't forget to compare it with analogue audio to see what it should sound like.

JamesH65, 2017-05-18:
@popcornmix @pelwell Should this remain open? There is a method of replicating.

JamesH65, 2019-01-18:
The link to the program that produces the error is broken, and this is nearly three years old.
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-07-30 by JamesH65