raspberrypi/linux #1625: CONFIG_TRIM_UNUSED_KSYMS in rpi-4.7.y default configs

clivem, 2016-08-30:
I'd think having CONFIG_TRIM_UNUSED_KSYMS enabled in the default configs in rpi-4.7.y is going to upset people trying to build out-of-tree modules. ;)

Ferroin, 2016-08-31:
It probably will, but it's indeterminate to what degree it will.  I('m kind of dubious of the value of this anyway, seeing as a huge number of drivers are built as modules, which means that it's likely trimming very few, if any, symbols (in which case it will have near zero impact now, but might in the future as the config and the kernel sources themselves change).

popcornmix, 2016-08-31:
I'll do a test just to see what the saving is, just to see if it is substantial.
But most likely this will be disabled.

clivem, 2016-09-14:
Whilst diffing my 4.7 configs against 4.4....

@popcornmix  I think these need to be enabled in 4.7 for parity with 4.4. 

CONFIG_SERIAL_SC16IS7XX_SPI=y https://github.com/raspberrypi/linux/commit/5ba1281cbd7f1d8db419f41919a51dedc4d5afa3
CONFIG_IPVLAN=m https://github.com/raspberrypi/linux/commit/5cb037ea9a01fd8b721842ba42bba86827fa5318
CONFIG_VXLAN=m https://github.com/raspberrypi/linux/commit/5ba1281cbd7f1d8db419f41919a51dedc4d5afa3
CONFIG_HID_GEMBIRD=m https://github.com/raspberrypi/linux/commit/40b1e4b9ba8d9450ce90007ca494bcf429d4f294

popcornmix, 2016-09-15:
Added - thanks.
Also removed CONFIG_TRIM_UNUSED_KSYMS

closed 2016-09-25 by clivem