raspberrypi/linux #1626: rpi-4.7.y brcmfmac sequence number error

clivem, 2016-08-31:
@pelwell You might want to merge the patch into rpi-4.7.y which "hides" the sequence number error. ;)

```
Feb 11 16:28:02 Akkordion3B1 kernel: Linux version 4.7.2-505.20160830git50500ec.fc24.armv7hl.bcm2709 (clivem@e5530) (gcc version 6.1.1 20160621 (Red Hat Cross 6.1.1-2) (GCC) ) #1 SMP Wed Aug 31 15:50:01 BST 2016

Feb 11 16:28:09 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24
Feb 11 16:28:09 Akkordion3B1 kernel: usbcore: registered new interface driver brcmfmac
Feb 11 16:28:09 Akkordion3B1 kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
Feb 11 16:28:09 Akkordion3B1 kernel: brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code (0x30 0x30)
Feb 11 16:28:11 Akkordion3B1 NetworkManager[325]: <info>  [1455208091.9890] rfkill0: found WiFi radio killswitch (at /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill0) (driver brcmfmac)
Feb 11 16:28:12 Akkordion3B1 kernel: brcmfmac: power management disabled
Feb 11 16:28:13 Akkordion3B1 kernel: brcmfmac: power management disabled
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 58: sequence number error, expect 57
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 57: sequence number error, expect 59
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 60: sequence number error, expect 59
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 63: sequence number error, expect 60
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 61: sequence number error, expect 64
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 67: sequence number error, expect 64
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 65: sequence number error, expect 68
Feb 11 16:28:21 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 69: sequence number error, expect 68
Aug 31 16:42:11 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 213: sequence number error, expect 211
Aug 31 16:42:11 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 211: sequence number error, expect 214
Aug 31 16:42:11 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 215: sequence number error, expect 214
Aug 31 16:42:11 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 253: sequence number error, expect 250
Aug 31 16:42:11 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 250: sequence number error, expect 254
Aug 31 16:42:11 Akkordion3B1 kernel: brcmfmac: brcmf_sdio_hdparse: seq 255: sequence number error, expect 254

```

clivem, 2016-09-03:
@pelwell Might be as well to cherry-pick these 3x commits....
https://github.com/raspberrypi/linux/commit/3b5ab11e2547168351370431208b7d903f4b2491
https://github.com/raspberrypi/linux/commit/25665b2105cf64006d66a9efcd32526bd4bf483d
https://github.com/raspberrypi/linux/commit/e9917292743ce5fd5bf4bb0e56c97e10cf18c743

popcornmix, 2016-09-03:
Okay added.

clivem, 2016-09-03:
@popcornmix Ta. Should stop people complaining about their logs being filled with those "out of sequence" messages.....

Totally the wrong place for this, but I'm just looking at overlays.... 
Overlays in 4.4.y but not 4.7.y .....
```
audremap.dtbo
mcp23s17.dtbo
sc16is750-i2c.dtbo
```
.... so we still have a few more 4.4.y patches to cherry-pick into 4.7.y. I'm out of time today, to look further....

popcornmix, 2016-09-03:
Added those (and a few more I found).

closed 2016-09-14 by clivem