raspberrypi/linux #1635: Opening /dev/ttyS0 with minicom hangs on 4.4.x

mypiandrew, 2016-09-12:
This relates to a custom compute module board, but I expect it could be reproduced on a standard Pi board 

in /boot/config.txt I have ttyS0 setup as follows

```
### UART1 ###
# Basic UART1 overlay with RX/TX pins only
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33

```

However doing the below hangs when opening ttyS0 via minicom

'# minicom -D /dev/ttyS0`

The 4.4.15/16/17/20 commits/kernels all have this problem

The earliest 4.4.x commit I tried was  4.4.12+ (which still had the problem)

Going back to the below fixed the issue :

 4.1.21+ (dea25fa62132365c11087e51e416df656db28bf3) 

If you need me to try and isolate this further let me know

pelwell, 2016-09-12:
Can you report the output of `raspi-gpio get`? If you don't have raspi-gpio you can apt-get install it.

mypiandrew, 2016-09-12:
Here you go : 

http://pastebin.com/DfEpzB7h

pelwell, 2016-09-12:
So those logs show that the pins are configured correctly in both cases, so the fault must lie elsewhere.

1) Could you post the output from dmesg (the failing case only should be enough)?

2) While minicom is running, can you `stty -F /dev/ttyS0 -a`? I'm wondering if the settings are incorrect. Again, just the bad build ought to be OK, but trying both and comparing might speed up the diagnosis.

mypiandrew, 2016-09-12:
Hi, I'm getting better (slowly) at this "hunt the patch" stuff (popcornmix is slowly teaching me!)

Will do it in reverse, do you want anything else like device tree stuff or anything from /proc ?

pelwell, 2016-09-12:
Ok, for completeness, the output of `dtc -I fs /proc/device-tree` will capture the whole Device Tree.

mypiandrew, 2016-09-12:
OK, this is strange

I'll eat humble pie a bit here in the sense that 4.4.15 (6ab8282964b77388e12cc7f5de7c3d7b72cdcf4c) is now working, so long as I comment out the CAN Bus overlay in config.txt (see pastebin link below)

On my system I can't fit both the comms card (that uses the ttyS0 pins) and the CAN BUS card so I guess that's OK.

But for what it's worth if I have the CAN BUS card installed and the overlay enabled, the CAN card gets detected AOK, but running : 

root@raspberrypi:~# minicom -D /dev/ttyS0

hangs the board totally

http://pastebin.com/wWCEx1Bd

mypiandrew, 2016-09-12:
root@raspberrypi:~# stty -F /dev/ttyS0 -a

also hangs the board when CAN overlay enabled

mypiandrew, 2016-09-12:
When CAN overlay installed :

dmesg:
http://pastebin.com/7xFC87qs

raspi-gpio get:
http://pastebin.com/wMENcbmi

devicetree :
http://pastebin.com/WkeF6ZGV

mypiandrew, 2016-09-12:
Interestingly, if I back off to 4.1.21 (dea25fa62132365c11087e51e416df656db28bf3) I can have the CAN overlay enabled and all works as expected with ttyS0

Does this help at all?

pelwell, 2016-09-12:
Yes, that helps a lot. The fact that you are using SPI as well is a vital bit of information, but you weren't to know that.

I think you are seeing the problem reported in https://github.com/raspberrypi/linux/issues/1484 (Aux SPI (SPI1) + Console on Aux UART (ttyS0) = Lockup). All the Aux peripherals share an interrupt, and that is thought to be the root of the lockups. Confirmation that the problem definitely didn't exist in the 4.1 tree and that somebody else is affected raises the priority of that issue.

I'm going to close this issue, but feel free to follow and contribute to #1484.

closed 2016-09-12 by pelwell

mypiandrew, 2016-09-12:
Oh wow, I'm not alone ;)
Thanks for confirming the problem, looks like it's an issue with the 4.4.x line the - will stick with 4.1.21 for now