raspberrypi/linux #1638: bootcode.bin fails with DHCP relay

jgfrm, 2016-09-15:
Created a SD-card with bootcode.bin cf https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md

If the PI is connected to the VLAN the physicial DHCP server is active on, the PI boots the kernel from the TFTP server.
However, if the PI is connected to another VLAN, that exposes DHCP services via DHCP relay, the PI can not obtain an IP address (checked with Wireshark).
It seems that bootcode.bin can not handle DHCP offers from a DHCP relay agent, but only from a DHCP server directly.

popcornmix, 2016-09-15:
Ping @ghollingworth

ghollingworth, 2016-09-15:
Do you have a pcap file of the DHCP request and reply (through the relay) so I can see what it'll look like?

jgfrm, 2016-09-15:
I have the output of two tcpdump traces.
The first one supposed a normal dhcp relay set up, in which the DHCP server identifier equals the IP address of the DHCP server.
I thought perhaps the PI was ignoring the DCHP offer because the PI is on a different subnet than the DHCP server.
Therefore, I created a second tcpdump that uses the router (relay) as the DHCP server identifier, This did not help.
I can email you the files if you like.

Best,
JG

ghollingworth, 2016-09-16:
Yes please gordon@raspberrypi.org

ghollingworth, 2016-09-16:
How have you set up the server, are you using dnsmasq with option 43 (vendor specific information containing "Raspberry Pi Boot   ")

Or are you putting the IP address of the tftp server in the TFTP server address option 66?

Gordon

jeffborg, 2016-09-26:
Hi I have the same issue here as well.

Got a 2nd switch with port mirroring to try and work out what's going on. The only differences between the 2 DHCP replies apart from the addresses are
A) The relay agent ip address is filled in
B) Hops are 1 instead of 0

I used a dhcpd with the same config as my router to test this

This response works
```
Bootstrap Protocol (Offer)
    Message type: Boot Reply (2)
    Hardware type: Ethernet (0x01)
    Hardware address length: 6
    Hops: 0
    Transaction ID: 0x26f30339
    Seconds elapsed: 0
    Bootp flags: 0x0000 (Unicast)
    Client IP address: 0.0.0.0
    Your (client) IP address: 192.168.11.101
    Next server IP address: 192.168.10.4
    Relay agent IP address: 0.0.0.0
    Client MAC address: Raspberr_e3:5e:2c (b8:27:eb:e3:5e:2c)
    Client hardware address padding: 00000000000000000000
    Server host name not given
    Boot file name: undionly.kpxe
    Magic cookie: DHCP
    Option: (53) DHCP Message Type (Offer)
        Length: 1
        DHCP: Offer (2)
    Option: (54) DHCP Server Identifier
        Length: 4
        DHCP Server Identifier: 192.168.11.2
    Option: (51) IP Address Lease Time
        Length: 4
        IP Address Lease Time: (86400s) 1 day
    Option: (43) Vendor-Specific Information
        Length: 20
        Value: 52617370626572727920506920426f6f74202020
    Option: (1) Subnet Mask
        Length: 4
        Subnet Mask: 255.255.255.0
    Option: (255) End
        Option End: 255
    Padding: 00000000000000000000000000000000
```

This one doesn't
```
Bootstrap Protocol (Offer)
    Message type: Boot Reply (2)
    Hardware type: Ethernet (0x01)
    Hardware address length: 6
    Hops: 1
    Transaction ID: 0x26f30339
    Seconds elapsed: 0
    Bootp flags: 0x0000 (Unicast)
    Client IP address: 0.0.0.0
    Your (client) IP address: 192.168.58.187
    Next server IP address: 192.168.10.4
    Relay agent IP address: 192.168.58.1
    Client MAC address: Raspberr_e3:5e:2c (b8:27:eb:e3:5e:2c)
    Client hardware address padding: 00000000000000000000
    Server host name not given
    Boot file name: undionly.kpxe
    Magic cookie: DHCP
    Option: (53) DHCP Message Type (Offer)
        Length: 1
        DHCP: Offer (2)
    Option: (54) DHCP Server Identifier
        Length: 4
        DHCP Server Identifier: 192.168.100.1
    Option: (51) IP Address Lease Time
        Length: 4
        IP Address Lease Time: (86400s) 1 day
    Option: (43) Vendor-Specific Information
        Length: 20
        Value: 52617370626572727920506920426f6f74202020
    Option: (1) Subnet Mask
        Length: 4
        Subnet Mask: 255.255.255.0
    Option: (255) End
        Option End: 255
    Padding: 00000000000000000000000000000000
```

I've also discovered that setting option 66 you don't need option 43 set.

I'm thinking if it's possible with dhcpd configuring a 2nd one on my subnet to not reply normally and only reply to the pi's boot request.

ghollingworth, 2016-09-26:
Yes the issue was the hops parameter, I've now fixed this, but not yet pushed...  Will do in the next couple of days...

Gordon

jeffborg, 2016-09-26:
I've spent hours on this, so I guess as this is baked into the pi3 :( I guess a small vm on the same subnet is in order to make this work.

ghollingworth, 2016-09-26:
Yes it is baked into the Pi, although using bootcode.bin on its own will enable this to work correctly (or should), but we need to test this because we do have the opportunity to take the fixes in bootcode.bin and apply them to a future revision of the SoC.

I should also note I've just added the ability for a gateway to be used for this mode as well...

If you can test this for me I'd be eternally grateful, please email me at gordon@raspberrypi.org and I'll send a bootcode.bin for you to try

jeffborg, 2016-10-18:
@ghollingworth any updates on this?

ghollingworth, 2016-10-18:
[bootcode.bin.zip](https://github.com/raspberrypi/linux/files/535799/bootcode.bin.zip)

Attached new bootcode.bin which you should be able to try out.  I'm in the process of pushing the fixes to next branch

Ruffio, 2017-02-05:
@jgfrm it looks like it has been resolved. Can this be closed?

jgfrm, 2017-02-05:
Solved

closed 2017-02-05 by jgfrm