raspberrypi/linux #1640: aarch64: rpi3: reboot doesn't work, hangs indefinitely

piso77, 2016-09-16:
After issuing a 'reboot', the board completes the shutdown process but hangs indefinitely:

[  OK  ] Stopped Load Kernel Modules.
[  OK  ] Unmounted /boot/firmware.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[ 4609.960504] reboot: Restarting system

To reboot i have to pull the usb cable everytime.
This only affect if the kernel is compiled in aarch64 mode, armhf is fine.

bcmrpi3_defconfig and gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.2).

brobwind, 2016-09-21:
Wow, you can compiled the kernel in aarch64 mode, great job!

popcornmix, 2016-09-21:
@Electron752 does reboot work for you?

Electron752, 2016-10-30:
Sorry, I didn't get any e-mail on this.

Yes, reboot works fine from both debian and raspbian.   I haven't tried ubuntu yet.

piso77, 2016-11-01:
I'm away from my board at the moment, next week i can give it another try.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-18 by JamesH65

kees, 2018-04-20:
I'm seeing this problem on aarch64 kernels too. Happens with both upstream kernel (at least up to 4.15) and Ubuntu kernel. Currently running:

4.15.0-15-generic #16-Ubuntu SMP Wed Apr 4 14:00:16 UTC 2018 aarch64 aarch64 aarch64 GNU/Linux

Updating to latest firmware (as of https://github.com/raspberrypi/firmware/commit/a154f2136850dba827cf4bc40794854376902cbd) didn't change anything.