raspberrypi/linux #1641: Set CONFIG_VMSPLIT_3G=y in kernel configuration will make the initramfs data lose

brobwind, 2016-09-18:
Hi, I am trying to make the newest Brillo system run on the Raspberry  Pi 2 Model B, In the Brillo source code master branch, it suggests add CONFIG_VMSPLIT_3G=y in the kernel configuration(https://android.googlesource.com/device/generic/brillo/+/5b23ce3ee5fecc9f40dad82b03990499237ba686%5E%21/#F0), but while I am trying to do this, the initramfs data in the main memory will lose(I need a initramfs file to boot):
The kernel have such messages:
<85>á[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.21-v7+ (hzak@B85RPI) (gcc version 4.9.x 20150123 (prerelease) (GCC) ) #2 SMP Sat Sep 17 16:07:49 CST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Truncating RAM at 0x00000000-0x3b000000 to -0x30000000
[    0.000000] Consider using a HIGHMEM enabled kernel.
[    0.000000] INITRD: 0x3aed7000+0x00118f60 is not a memory region - disabling initrd
[    0.000000] cma: Reserved 8 MiB at 0x2f400000
[    0.000000] Memory policy: Data cache writealloc

From these logs, I known: 
- The RAM is truncated to 0x30000000
- The initrd has been disabled due to locate in 0x3aed700+0x00118f60
- To enable HIGHMEM configuration

But while I trying to enable CONFIG_HIGHMEM in kernel, it will cause another kernel crash:
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.21-v7+ (hzak@B85RPI) (gcc version 4.9.x 20150123 (prerelease) (GCC) ) #2 SMP Sun Sep 18 14:11:00 CST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (101620->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 11 pages/cpu @ef768000 s23104 r0 d21952 u45056
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239936
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa21041 bcm2709.serial=0x35d1198a smsc95xx.macaddr=B8:27:EB:D1:19:8A bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 rootwait noinitrd init=/init elevator=deadline androidboot.hardware=rpi androidboot.selinux=permissive androidboot.disk.boot=/dev/block/mmcblk0p1 androidboot.disk.system=/dev/null androidboot.disk.data=/dev/null androidboot.mode=recovery initcall_debug=1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 933952K/966656K available (9865K kernel code, 369K rwdata, 1368K rodata, 1024K init, 906K bss, 24512K reserved, 8192K cma-reserved, 171104K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0bf86c8   (12226 kB)
[    0.000000]       .init : 0xc0c00000 - 0xc0d00000   (1024 kB)
[    0.000000]       .data : 0xc0d00000 - 0xc0d5c450   ( 370 kB)
[    0.000000]        .bss : 0xc0dc70dc - 0xc0ea99d4   ( 907 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000012] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000032] Switching to timer-based delay loop, resolution 52ns
[    0.000362] Console: colour dummy device 80x30
[    0.000412] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000440] pid_max: default: 32768 minimum: 301
[    0.000600] Security Framework initialized
[    0.000622] SELinux:  Initializing.
[    0.000983] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001006] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002471] Disabling cpuset control group subsystem
[    0.002525] Initializing cgroup subsys io
[    0.002564] Initializing cgroup subsys memory
[    0.002630] Initializing cgroup subsys devices
[... 136 lines cut ...]
[    1.568495] [<c010f88c>] (secondary_start_kernel) from [<001016ac>] (0x1016ac)
[    1.575719] CPU2: stopping
[    1.578436] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D         4.4.21-v7+ #2
[    1.585917] Hardware name: BCM2709
[    1.589337] [<c0111f80>] (unwind_backtrace) from [<c010d174>] (show_stack+0x20/0x24)
[    1.597093] [<c010d174>] (show_stack) from [<c04a385c>] (dump_stack+0xc8/0x120)
[    1.604413] [<c04a385c>] (dump_stack) from [<c010fe5c>] (handle_IPI+0x2c8/0x31c)
[    1.611821] [<c010fe5c>] (handle_IPI) from [<c01015c4>] (bcm2836_arm_irqchip_handle_irq+0x7c/0xbc)
[    1.620790] [<c01015c4>] (bcm2836_arm_irqchip_handle_irq) from [<c010df38>] (__irq_svc+0x58/0x78)
[    1.629663] Exception stack(0xef13bf30 to 0xef13bf78)
[    1.634720] bf20:                                     00000000 ef77e388 c0d044b4 00000000
[    1.642907] bf40: c0d0359c c0100000 c0d586e4 c0d9bb48 c0d0359c c0c9f364 00000000 ef13bf8c
[    1.651091] bf60: ef13bf80 ef13bf80 c0108e80 c0108e84 60000013 ffffffff
[    1.657716] [<c010df38>] (__irq_svc) from [<c0108e84>] (arch_cpu_idle+0x34/0x54)
[    1.665126] [<c0108e84>] (arch_cpu_idle) from [<c0168aec>] (default_idle_call+0x34/0x48)
[    1.673230] [<c0168aec>] (default_idle_call) from [<c0168d8c>] (cpu_startup_entry+0x28c/0x338)
[    1.681854] [<c0168d8c>] (cpu_startup_entry) from [<c010f88c>] (secondary_start_kernel+0x16c/0x194)
[    1.690908] [<c010f88c>] (secondary_start_kernel) from [<001016ac>] (0x1016ac)hzak@B85RPI:~$

NOTE:
- firmware version: 8979042
- kernel version: 2d31cd5
- initramfs configured in boot partition's config.txt

Thanks!

pelwell, 2016-09-20:
The firmware is expecting the kernel to be configured with a 2+2 split so that it can address all available RAM without using HIGHMEM (which we couldn't get to work). Even with HIGHMEM enabled I'm not sure it will work early enough in the boot for initramfs.

Try using the `ramfsaddr` config.txt setting to load the initramfs file to a lower address - 0x2f000000 or lower ought to work, e.g. `ramfsaddr=0x2f000000`, or even lower `ramfsaddr=0x10000000`.

brobwind, 2016-09-20:
Yes, Without using HIGHMEM, it does work when set ramfsaddr=0x2f000000 or ramfsaddr=0x10000000, or using initramfs ramdisk7.img 0x2f000000. I think it's a little dangerous since I don't known the kernel loading address(loading rule) and the reserved video memory address(the RAM layout seems not document yet).
I just hope this problem can be solved.
Thanks

pelwell, 2016-09-20:
The kernel is always loaded to 0x8000, so setting the address as you have done is safe. However, it is a bit hacky.

As it turns out, we've already restricted the load address for the device tree blob to below 0x30000000 (unless hard coded by the user or the loader stub), and I don't see any reason why we can't do that for the initramfs blob as well - it's just that nobody's asked before.

pelwell, 2016-09-20:
Can you try this [test start.elf](https://drive.google.com/file/d/0B_P-i4u-SLBXb0RZUGFUM1hWTFU/view?usp=sharing) for me? It is hard-coded to have gpu_mem set to 128K, but that should be OK for testing.

To use it, make a copy of your existing /boot/start.elf then copy this one in its place and reboot, having first ensured that you haven't set `start_x=1` or `start_debug=1`. If that boots OK for you, try removing the explicit initramfs address (use `followkernel` instead), then report back.

Assuming there are no problems I'll get the relevant patch into the next firmware release.

brobwind, 2016-09-21:
Just let you known, configure kernel with VMSPLIT_3G and without HIGHMEM, and set initramfs to ramdisk7.img followkernel, the initramfs still locate above 0x30000000(with followkernel, the initramfs address should locate at the end of kernel?), here is the kernel message:
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.21-v7+ (hzak@B85RPI) (gcc version 4.9.x 20150123 (prerelease) (GCC) ) #1 SMP Tue Sep 20 22:16:55 CST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
**[    0.000000] Truncating RAM at 0x00000000-0x3b000000 to -0x30000000**
[    0.000000] Consider using a HIGHMEM enabled kernel.
**[    0.000000] INITRD: 0x3af08000+0x000e74ee is not a memory region - disabling initrd**
[    0.000000] cma: Reserved 8 MiB at 0x2f400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (101620->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 11 pages/cpu @effb2000 s23104 r0 d21952 u45056
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa21041 bcm2709.serial=0x35d1198a smsc95xx.macaddr=B8:27:EB:D1:19:8A bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 rootwait noinitrd init=/init elevator=deadline androidboot.hardware=rpi androidboot.selinux=permissive androidboot.disk.boot=/dev/block/mmcblk0p1 androidboot.disk.system=/dev/null androidboot.disk.data=/dev/null androidboot.mode=recovery initcall_debug=1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 756248K/786432K available (9865K kernel code, 367K rwdata, 1368K rodata, 1024K init, 888K bss, 21992K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0bf84d4   (12226 kB)
[    0.000000]       .init : 0xc0c00000 - 0xc0d00000   (1024 kB)
[    0.000000]       .data : 0xc0d00000 - 0xc0d5bf90   ( 368 kB)
[    0.000000]        .bss : 0xc0dc709c - 0xc0ea5114   ( 889 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16

I will test the new start.elf later.
Thanks

brobwind, 2016-09-21:
After using the new start.elf with VMSPLIT_3G and without HIGHMEM, the kernel seems can not boot(there is not kernel message from the serial port), with follow configurations:
- initramfs ramdisk7.img
- initramfs ramdisk7.img 0x2f000000
- initramfs ramdisk7.img followkernel
NOTE: I always do not use the mkknlimg script just copy the arch/arm/boot/zImage.
Thanks

brobwind, 2016-09-21:
All related files can download from https://github.com/brobwind/rpi2b-kernel-issue
firmware version: 8979042 ( 2016-09-15 14:23:54)
boot/kernel7.img and kernel7.img-3g are the same and can boot
kernel7.img-3g-highmem can not boot
Thanks

pelwell, 2016-09-21:
Thanks - that was helpful. It seems that loading just below 0x30000000 causes problems, so I've updated the [test start.elf](https://drive.google.com/file/d/0B_P-i4u-SLBXb0RZUGFUM1hWTFU/view?usp=sharing) to load below 0x2f000000 instead. With that I get serial port output, but I think it then stalls because the rest of my SD card isn't correct.

Try the updated start.elf yourself - I hope it will work for you.

brobwind, 2016-09-21:
It seems these two test start.elf are the same(with the same link address). And have the same sha1sum after downloaded, please help to check again.
Thanks

pelwell, 2016-09-21:
The original file had the sha1sum of 90a57a705bdd99f14d859d9f94d1a424e948862f. The new one has a sha1sum of 0a2b6b08bf2ad95e1bb3d81fb8178d0b86a9d800. The link didn't change because I'm using the Google Drive versioning system.

brobwind, 2016-09-21:
Sorry, my mistake. But the new start.elf(0a2b6b0) still can not boot the kernel, no serial port output. After replace with the old one (firmware @ 8979042), it has the serial port output.

pelwell, 2016-09-21:
I started with a clean Raspbian installation, scratched the boot partition and copied your files in then overwrote the start.elf. For some reason it wasn't booting, except when started from our debugger. I've done a clean build, copied it to the card and it boots (to a point, with serial output) from cold, followkernel and an explit 0x2f000000. That version I've now uploaded (direct from the SD card) to the same URL.

If that doesn't at least get you some serial output then I'm stumped.

brobwind, 2016-09-22:
Great job, it can boot now with following configurations:
- initramfs ramdisk7.img: initrd locate at 0x2ef18000 ~ 0x2efff4ee
- initramfs ramdisk7.img followkernel: initrd also locate at  0x2ef18000 ~ 0x2efff4ee (I am really confused, as you mentioned previously default kernel image start address is 0x8000, the initrd start address should not stay such high).

Thanks

pelwell, 2016-09-22:
One of the first things the kernel does is to read and internalise the Device Tree blob, after which it no longer needs the memory it occupied. The same happens with initramfs a short while later:
```
[    0.312126] Trying to unpack rootfs image as initramfs...
[    0.383790] Freeing initrd memory: 928K (eef18000 - ef000000)
```
So I don't think it matters where we load the DTB and initramfs to, provided it isn't too high (as we've seen) or so low that it collides with the end of the kernel.

With a Pi3 you will need `enable_uart=1` if you want serial port output, but you don't need to set the `gpu_mem` so high any more. The reason it used to help is that it only left 768MB for the kernel, all of which is addressable as LOWMEM. (Yes, 1024-240 is not 768, but the top 16MB is effectively unusable, so the calculation is really 1008 - 240 = 768).

Now that the automatic DTB+initramfs placement is limited to the first 768MB you can run with any `gpu_mem` size you like, or at least you will once we've made a proper firmware release containing this change.

brobwind, 2016-09-22:
You are right. Thank you very much

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65