raspberrypi/linux #1642: How to build my own security module based on lsm?

zfyseu, 2016-09-18:
Hello,everyone!

I want build my own security module based on lsm on Rpi 2B.
And the following is my code:
geek_lsm.c:

`#include<linux/security.h>
  #include<linux/sysctl.h>
  static unsigned long long count = 0;
  int task_create_hook(unsigned long clone_flags)
  {
  	printk("[+geek] call task_create(). count=%llu\n", ++count);	
  	return 0;
  }
  
  static struct security_operations geek_ops = {
  	.name = "geek",
  	.task_create = task_create_hook,
  };
  
  static __init int geek_init(void)
  {
  	printk("[+geek] loading...\n");	
  	if(register_security(&geek_ops)){
  		printk("[+geek] register faild\n");	
  	}
  	return 0;
  }
  
  security_initcall(geek_init);`

However, when I build the kernel, it return with the error:
`security/geek/geek_lsm.c: At top level:
security/geek/geek_lsm.c:72:17: error: variable ‘geek_ops’ has initializer but incomplete type
   static struct security_operations geek_ops = {
                 ^
security/geek/geek_lsm.c:73:4: error: unknown field ‘name’ specified in initializer
    .name = "geek",
    ^
security/geek/geek_lsm.c:73:4: warning: excess elements in struct initializer [enabled by default]
security/geek/geek_lsm.c:73:4: warning: (near initialization for ‘geek_ops’) [enabled by default]
security/geek/geek_lsm.c:74:4: error: unknown field ‘task_create’ specified in initializer
    .task_create = task_create_hook,
    ^
security/geek/geek_lsm.c:74:4: warning: excess elements in struct initializer [enabled by default]
security/geek/geek_lsm.c:74:4: warning: (near initialization for ‘geek_ops’) [enabled by default]
security/geek/geek_lsm.c: In function ‘geek_init’:
security/geek/geek_lsm.c:80:4: error: implicit declaration of function ‘register_security’ [-Werror=implicit-function-declaration]
    if(register_security(&geek_ops)){
    ^
cc1: some warnings being treated as errors
make[2]: *** [security/geek/geek_lsm.o] Error 1
make[1]: *** [security/geek] Error 2
make: *** [security] Error 2
`
Is there anyone know how to deal with the problem?

And I don't find the defination of the struct "security_operations" in the file "/linux/include/linux/security.h"

6by9, 2016-09-18:
It looks like security_operations in security.h moved around a lot in the 4.2 release. Initially to include/linux/lsm_hooks.h, and then was removed with commit 
```
 commit b1d9e6b0646d0e5ee5d9050bd236b6c65d66faef
Author: Casey Schaufler <casey@schaufler-ca.com>
Date:   Sat May 2 15:11:42 2015 -0700

    LSM: Switch to lists of hooks
    
    Instead of using a vector of security operations
    with explicit, special case stacking of the capability
    and yama hooks use lists of hooks with capability and
    yama hooks included as appropriate.
    
    The security_operations structure is no longer required.
    Instead, there is a union of the function pointers that
    allows all the hooks lists to use a common mechanism for
    list management while retaining typing. Each module
    supplies an array describing the hooks it provides instead
    of a sparsely populated security_operations structure.
    The description includes the element that gets put on
    the hook list, avoiding the issues surrounding individual
    element allocation.
    
    The method for registering security modules is changed to
    reflect the information available. The method for removing
    a module, currently only used by SELinux, has also changed.
    It should be generic now, however if there are potential
    race conditions based on ordering of hook removal that needs
    to be addressed by the calling module.
    
    The security hooks are called from the lists and the first
    failure is returned.
```
You'll need to read up as to the correct way of doing this under kernel 4.4 and above.

zfyseu, 2016-09-19:
Thanks,6by9!
I have solved the problem according to your advice.

the following is the modified code:

#include<linux/lsm_hooks.h>
#include ...
...

static struct security_hook_list geek_ops[] = {
         LSM_HOOK_INIT(task_create, task_create_hook),
  };

  void __init  geek_init(void)
  {
        printk("[+geek] loading...\n");
        security_add_hooks(geek_ops, ARRAY_SIZE(geek_ops));

  }

 security_initcall(geek_init);

closed 2016-09-19 by zfyseu