raspberrypi/linux #1643: raspberrypi 3b wlan can not find

zhangsanshi, 2016-09-18:
I'm interested in pi, and thanks for yours work.
I have difficult while using it. 
I run the command `rpi-update` and * I touch it with some water *.
After reboot, pi can't find `wlan`, see it  in the below command.
I want to know how to check the pi and find the real reason.

```
lsusb
Bus 001 Device 004: ID 1307:0165 Transcend Information, Inc. 2GB/4GB Flash Drive
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
           Chassis: n/a
        Machine ID: 18634e1404184594bc66ddacfba0bf43
           Boot ID: e4f6b1579a70476f8cdca6f58869095a
  Operating System: Raspbian GNU/Linux 8 (jessie)
            Kernel: Linux 4.4.21-v7+
      Architecture: arm
```

```
lsmod
Module                  Size  Used by
```

```
uname -a
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux
```

pelwell, 2016-09-18:
The Pi3 built-in WLAN adapter isn't a USB device - it is an SDIO device connected to the second SD/MMC interface.

Can you post or link to the whole output of dmesg after booting?

zhangsanshi, 2016-09-19:
```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.21-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #911 SMP Thu Sep 15 14:22:38 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 808c2f40, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f62000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa22082 bcm2709.serial=0x54cf19d1 smsc95xx.macaddr=B8:27:EB:CF:19:D1 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/sda2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939072K/966656K available (6348K kernel code, 432K rwdata, 1716K rodata, 476K init, 764K bss, 19392K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807e854c   (8066 kB)
      .init : 0x807e9000 - 0x80860000   ( 476 kB)
      .data : 0x80860000 - 0x808cc250   ( 433 kB)
       .bss : 0x808cf000 - 0x8098e1ec   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000284] Console: colour dummy device 80x30
[    0.001338] console [tty1] enabled
[    0.001383] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001453] pid_max: default: 32768 minimum: 301
[    0.001789] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001832] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002791] Disabling cpuset control group subsystem
[    0.002847] Initializing cgroup subsys io
[    0.002898] Initializing cgroup subsys memory
[    0.002963] Initializing cgroup subsys devices
[    0.003005] Initializing cgroup subsys freezer
[    0.003047] Initializing cgroup subsys net_cls
[    0.003118] CPU: Testing write buffer coherency: ok
[    0.003202] ftrace: allocating 21220 entries in 63 pages
[    0.053031] CPU0: update cpu_capacity 1024
[    0.053096] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.053130] [bcm2709_smp_prepare_cpus] enter
[    0.053285] Setting up static identity map for 0x8240 - 0x8274
[    0.054933] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.055150] [bcm2709_secondary_init] enter cpu:1
[    0.055192] CPU1: update cpu_capacity 1024
[    0.055199] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.055570] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.055734] [bcm2709_secondary_init] enter cpu:2
[... 243 lines cut ...]
[   14.724880] systemd-journald[137]: Received request to flush runtime journal from PID 1
[   15.081264] uart-pl011 3f201000.uart: no DMA platform data
[   15.333420] Adding 102396k swap on /var/swap.  Priority:-1 extents:3 across:200700k FS
[   15.643039] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   17.193376] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
```

```
ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:cf:19:d1  
          inet addr:192.168.31.188  Bcast:192.168.31.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4802 errors:0 dropped:2 overruns:0 frame:0
          TX packets:2452 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:334619 (326.7 KiB)  TX bytes:541939 (529.2 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:164 errors:0 dropped:0 overruns:0 frame:0
          TX packets:164 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:14708 (14.3 KiB)  TX bytes:14708 (14.3 KiB)
```

pelwell, 2016-09-19:
That shows that the Pi is finding the SDIO device but not using it. Try this command than upload or post  mmcinfo.txt:
```
(cd /sys/class/mmc_host/mmc1/mmc1:0001; find . -type f -exec grep -H . {} \; >& mmcinfo.txt)
```

zhangsanshi, 2016-09-19:
```
./type:SDIO
./power/control:auto
./power/runtime_active_time:0
grep: ./power/autosuspend_delay_ms: Input/output error
./power/runtime_status:unsupported
./power/runtime_suspended_time:0
./mmc1:0001:1/class:0x00
./mmc1:0001:1/power/control:auto
./mmc1:0001:1/power/runtime_active_time:0
grep: ./mmc1:0001:1/power/autosuspend_delay_ms: Input/output error
./mmc1:0001:1/power/runtime_status:unsupported
./mmc1:0001:1/power/runtime_suspended_time:0
./mmc1:0001:1/modalias:sdio:c00v02D0dA9A6
./mmc1:0001:1/device:0xa9a6
./mmc1:0001:1/uevent:SDIO_CLASS=00
./mmc1:0001:1/uevent:SDIO_ID=02D0:A9A6
./mmc1:0001:1/uevent:MODALIAS=sdio:c00v02D0dA9A6
./mmc1:0001:1/vendor:0x02d0
./mmc1:0001:2/class:0x00
./mmc1:0001:2/power/control:auto
./mmc1:0001:2/power/runtime_active_time:0
grep: ./mmc1:0001:2/power/autosuspend_delay_ms: Input/output error
./mmc1:0001:2/power/runtime_status:unsupported
./mmc1:0001:2/power/runtime_suspended_time:0
./mmc1:0001:2/modalias:sdio:c00v02D0dA9A6
./mmc1:0001:2/device:0xa9a6
./mmc1:0001:2/uevent:SDIO_CLASS=00
./mmc1:0001:2/uevent:SDIO_ID=02D0:A9A6
./mmc1:0001:2/uevent:MODALIAS=sdio:c00v02D0dA9A6
./mmc1:0001:2/vendor:0x02d0
./uevent:MMC_TYPE=SDIO
./uevent:MMC_NAME=
./uevent:MODALIAS=mmc:block
```

pelwell, 2016-09-19:
Interesting - what seems to be missing compared to a Pi 3 with working WiFi is any mention of the the driver:
```
./mmc1:0001:1/uevent:DRIVER=brcmfmac_sdio
...
./mmc1:0001:2/uevent:DRIVER=brcmfmac_sdio
```
What do you get from:
```
grep sdio /lib/modules/`uname -r`/modules.alias
```
?

zhangsanshi, 2016-09-20:
```
grep: /lib/modules/4.4.21-v7+/modules.alias: No such file or directory
```

zhangsanshi, 2016-09-20:
```
/opt/vc/bin/vcgencmd version 
Sep 14 2016 19:56:31 
Copyright (c) 2012 Broadcom
version efa16f891edfcb0ba1aae15be195ad9d8b38d6be (tainted) (release)
```

pelwell, 2016-09-20:
Something has gone wrong there - that directory should exist.

1) Is your SD card full? (`df /`) If so, you can delete any old /lib/modules/<version> directories.

2) If the card isn't full, there may have been a glitch with the previous update. Try running `sudo rpi-update` again and watch for any error messages.

zhangsanshi, 2016-09-21:
Thank you very much. the  `wlan` come back.
You give me a lot of help.

pelwell, 2016-09-21:
To help anybody else in the same situation, was the card full?

zhangsanshi, 2016-09-21:
I run `sudo rpi-update` and solution to the problem. 
I also get the error about full card at the downloading `rpi-4.4-y` form github.
I remove the apt cache, then `sudo rpi-update` success.

pelwell, 2016-09-21:
Thanks. Please close the issue if you are satisfied.

closed 2016-09-21 by zhangsanshi