raspberrypi/linux #1644: HDMI mode settings in config.txt are ignored when experimental GL driver is enabled.

camthesaxman, 2016-09-18:
My setup is a Raspberry Pi 3 Model B, and I am using a HDMI to VGA adapter to connect to a LCD monitor.
When booting a fresh install of Raspbian, the monitor resolution is correctly identified as 1024x768, and both the console and X display correctly in that resolution. When the GL driver is enabled, I no longer see the console at all (monitor says "No signal"), and X displays in 1152x864 (Monitor says "Out of timing. V:60Hz H:53.7kHz). I would not expect enabling an OpenGL driver to screw up the screen resolution. I tried to correct this by adding the following lines to config.txt, but they have no effect if the GL driver is enabled.
```
# Set width (in pixels) of text console
framebuffer_width=1024
#Set height (in pixels) of text console
framebuffer_height=768
# Set monitor mode to DMT
hdmi_group=2
# Set monitor resolution to 1024x768 XGA 60Hz (HDMI_DMT_XGA_60)
hdmi_mode=16
```

6by9, 2016-09-18:
config.txt video mode settings only apply to the firmware blob. OpenGL does not use it, and will not use it as it is using the standard Linux mechanisms. You're probably looking for xrandr instead.

camthesaxman, 2016-09-18:
Ok. What do you mean by standard Linux mechanisms? I would expect to get the same behavior with regards to screen resolution whether I use the default llvmpipe driver or the experimental driver. Since the text console does not use OpenGL, why is it affected? Is there a bug with KMS?

6by9, 2016-09-19:
Under the old scheme, the GPU sets up the hardware to compose the screen output. If the kernel requests it then that includes the frame buffer. This is why you use config.txt and commands such as tvservice to talk to the GPU to set/get parameters

Under the upstream driver (which is more than just OpenGL), the Linux kernel is responsible for ALL composition, including the frame buffer, therefore the config needs to be passed to that driver.

All bugs relating to the upstream driver should be posted to the upstream repo at https://github.com/anholt/linux. This repo pulls from there and does not include any of the development effort.
There is also a bug reporting thread on the forums - https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=135790

popcornmix, 2016-09-19:
@camthesaxman 
This may help: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=159853

But as @6by9 says config.txt applies to the firmware display (kms).
Using ``vc4-kms-v3d`` means the firmware display driver is not used and you can configure using standard arm-side linux tools.

But the arm-side kms driver is still experimental and may have issues. Using the ``vc4-fkms-v3d`` driver allows the experimental arm-side 3d driver to be used with the firmware display driver.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-18 by JamesH65

SqTER-PL, 2019-07-25:
I have the same problem and I found a solution.
Enter via SSH "tvservice -p"
The command without parameters will set the VODEO MODE stored in config.txt
You can add to autostart.