raspberrypi/linux #1649: watchdag caused traceback due to USB hang

andersthomson, 2016-09-22:
Hi,

I've pi2 used as a headless tvheadend server with two tuners (so lots of USB traffic). Every so often (after a day or two) I get hard hangs of all USB traffic which has prompted me to manually reboot the thing.

I've now enabled the watchdog, configured reboot on panic and gotten myself a serial cable and I now start to see results.

Attached is a large log of where the USB goes south and eventually I get a kernel backtrace triggered by a watchdog codepath somehow.

Any ideas how to fix this? 
(grep for  'cut here' for the kernel bits)

[crashreport.txt](https://github.com/raspberrypi/linux/files/488229/crashreport.txt)

popcornmix, 2016-09-24:
How did you enable the watchdog?
This look suspicious
```
Sep 22 18:06:13 rpi2 systemd-shutdow[1]: Failed to set timeout to 600s: Invalid argument
```
The hardware watchdog is limited to around 30 seconds I believe due the number of bits in the hardware countdown timer. Ten minutes is very long for a watchdog timeout. 10 seconds would be a more common value.

lategoodbye, 2016-09-24:
@andersthomson 
Please provide a lsusb -t output of the affected setup.
Are you able to narrow down the affected setup to a minimum setup of connected USB devices?

andersthomson, 2016-09-25:
watchdog is enabled as with the systemd unit, which exec's /usr/sbin/watchdog

 cat /etc/watchdog.conf 
#ping                   = 172.31.14.1
#ping                   = 192.168.1.1
#ping                   = 172.26.1.255
#interface              = eth0
#file                   = /var/log/messages
#change                 = 1407

# Uncomment to enable test. Setting one of these values to '0' disables it.
# These values will hopefully never reboot your machine during normal use
# (if your machine is really hung, the loadavg will go much higher than 25)
max-load-1              = 24
#max-load-5             = 18
#max-load-15            = 12

# Note that this is the number of pages!
# To get the real size, check how large the pagesize is on your machine.
min-memory              = 1
allocatable-memory      = 1

#repair-binary          = /usr/sbin/repair
#repair-timeout         = 
#test-binary            = 
#test-timeout           = 

watchdog-device = /dev/watchdog
watchdog-timeout = 15

# Defaults compiled into the binary
#temperature-device     =
temperature-device = /sys/class/thermal/thermal_zone0/temp
#max-temperature        = 120
max-temperature = 51000

# Defaults compiled into the binary
admin                   = root
interval                = 1
logtick                = 1
log-dir         = /var/log/watchdog

# This greatly decreases the chance that watchdog won't be scheduled before
# your machine is really loaded
realtime                = yes
priority                = 1

# Check if rsyslogd is still running by enabling the following line
#pidfile                = /var/run/rsyslogd.pid

andersthomson, 2016-09-25:
lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 2: Dev 6, If 0, Class=Vendor Specific Class, Driver=em28xx, 480M
        |__ Port 3: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 7, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 1: Dev 9, If 0, Class=Vendor Specific Class, Driver=em28xx, 480M
                |__ Port 2: Dev 11, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
            |__ Port 4: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 1: Dev 10, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 4: Dev 12, If 0, Class=Mass Storage, Driver=usb-storage, 480M

andersthomson, 2016-09-25:
My "is things in a stable state?" script detected this in dmesg yesterday, which also points to the watchdog.

[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.15-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #897 SMP Tue Jul 12 18:42:55 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 808c0e00, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f63000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa01041 bcm2709.serial=0x9aa48012 smsc95xx.macaddr=B8:27:EB:A4:80:12 bcm2708_fb.fbswap=1 bcm2709.uart_clock=3000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  root=/dev/mmcblk0p4 smsc95xx.turbo_mode=N rootdelay=2 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 init=/usr/lib/systemd/systemd ro
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939088K/966656K available (6344K kernel code, 432K rwdata, 1712K rodata, 476K init, 764K bss, 19376K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x807e6470   (8058 kB)
                     .init : 0x807e7000 - 0x8085e000   ( 476 kB)
                     .data : 0x8085e000 - 0x808ca108   ( 433 kB)
                      .bss : 0x808cd000 - 0x8098c1ac   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000011] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000031] Switching to timer-based delay loop, resolution 52ns
[    0.000349] Console: colour dummy device 80x30
[    0.000386] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000412] pid_max: default: 32768 minimum: 301
[    0.000802] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000824] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002003] Disabling cpuset control group subsystem
[    0.002049] Initializing cgroup subsys io
[    0.002085] Initializing cgroup subsys memory
[    0.002140] Initializing cgroup subsys devices
[    0.002166] Initializing cgroup subsys freezer
[    0.002192] Initializing cgroup subsys net_cls
[    0.002262] CPU: Testing write buffer coherency: ok
[    0.002348] ftrace: allocating 21209 entries in 63 pages
[    0.054644] CPU0: update cpu_capacity 1024
[    0.054683] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.054699] [bcm2709_smp_prepare_cpus] enter
[    0.054835] Setting up static identity map for 0x8240 - 0x8274
[    0.057213] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.057643] [bcm2709_secondary_init] enter cpu:1
[    0.057702] CPU1: update cpu_capacity 1024
[    0.057710] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.058283] [bcm2709_boot_secondary] cpu:2 started (0) 17
[    0.058636] [bcm2709_secondary_init] enter cpu:2
[    0.058668] CPU2: update cpu_capacity 1024
[... 388 lines cut ...]
[60700.429371] [<80029630>] (irq_exit) from [<800710a8>] (__handle_domain_irq+0x70/0xc4)
[60700.429391] [<800710a8>] (__handle_domain_irq) from [<80010a74>] (handle_IRQ+0x2c/0x30)
[60700.429408] [<80010a74>] (handle_IRQ) from [<8000951c>] (bcm2836_arm_irqchip_handle_irq+0xb8/0xbc)
[60700.429430] [<8000951c>] (bcm2836_arm_irqchip_handle_irq) from [<805b9684>] (__irq_svc+0x44/0x5c)
[60700.429437] Exception stack(0x8085ff08 to 0x8085ff50)
[... cut ...]
[64471.501568] em28178 #0: writing to i2c device at 0xc8 failed (error=-5)
[64472.101881] em28178 #0: writing to i2c device at 0xc8 failed (error=-5)
[64472.121581] em28174 #1: writing to i2c device at 0xd8 failed (error=-5)
[64472.128889] i2c i2c-6: cxd2820r: i2c rd failed=-5 reg=10 len=1
[64472.214369] em28178 #0: write to i2c device at 0xc8 failed with unknown error (status=3)
[64473.167852] em28174 #1: write to i2c device at 0xd8 failed with unknown error (status=1)
[64473.176946] i2c i2c-6: cxd2820r: i2c rd failed=-5 reg=10 len=1

P33M, 2016-09-27:
You appear to have 4 (?!) USB hubs of various sizes plugged into the Pi. Can you post the output of `lsusb -t` please?

Can you also say whether these hubs are self-powered or bus-powered?

andersthomson, 2016-09-28:
[lsusb.txt](https://github.com/raspberrypi/linux/files/497132/lsusb.txt)

Two self powered hubs off the pi2, one of them is a 10 port with carries three 4way splitters internally.

I got the hubs in an attempt to rule out power issues as I have to (supposedly power hungry) TV tuner sticks in there.

P33M, 2017-05-04:
Can you try again after running rpi-update? There have been some bug fixes that may affect your use case.

andersthomson, 2017-05-14:
I'm compiling my own stuff. Can you indicate git commit and config to use?

Thanks!
/Anders

popcornmix, 2017-05-14:
He'll be referring to top of rpi-4.9.y tree (currently https://github.com/raspberrypi/linux/commit/9a5f215eda12bad29b35040dff00d0346fe517e2) and using the bcm2709_defconfig.
See: https://www.raspberrypi.org/documentation/linux/kernel/building.md

But it is more useful for us if you test with a default kernel.
Reporting a bug with a custom built kernel adds another level of unknown when we try to diagnose.

andersthomson, 2017-05-14:
> He'll be referring to top of rpi-4.9.y tree (currently raspberrypi/linux@
> 9a5f215

Building that commit against that config right now. As it happens, I'm on
rpi-4.9.17 right now, and cannot recall seeing the Watchdog in the oops for
quite some time. I do still get regular oopes from the usb otg driver
though (and an attempt at vanilla-4.11 with dwc2 consistently fails to get
the tuners operational)

lategoodbye, 2017-05-14:
@andersthomson Could you please explain what goes wrong on dwc2?

andersthomson, 2017-05-15:
So, for what its worth. Freshly compiled. Not sure where this oops is
pointing.

/Anders

[   15.761953] EXT4-fs (dm-7): mounted filesystem with ordered data mode.
Opts: (null)
[   17.237677] EXT4-fs (dm-4): mounted filesystem with ordered data mode.
Opts: (null)
[   18.265822] bridge: filtering via arp/ip/ip6tables is no longer
available by default. Update your scripts to load br_netfilter if you need
this.
[   18.386555] IPv6: ADDRCONF(NETDEV_UP): br0: link is not ready
[   18.402539] IPv6: ADDRCONF(NETDEV_UP): veth-osc: link is not ready
[   18.417437] br0: port 1(veth-osc-br) entered blocking state
[   18.417446] br0: port 1(veth-osc-br) entered disabled state
[   18.418049] device veth-osc-br entered promiscuous mode
[   18.423906] IPv6: ADDRCONF(NETDEV_UP): veth-tvh: link is not ready
[   18.430086] br0: port 2(veth-tvh-br) entered blocking state
[   18.430095] br0: port 2(veth-tvh-br) entered disabled state
[   18.430869] device veth-tvh-br entered promiscuous mode
[   18.438490] br0: port 3(eth0) entered blocking state
[   18.438498] br0: port 3(eth0) entered disabled state
[   18.439088] device eth0 entered promiscuous mode
[   18.456215] br0: port 1(veth-osc-br) entered blocking state
[   18.456228] br0: port 1(veth-osc-br) entered forwarding state
[   18.456559] IPv6: ADDRCONF(NETDEV_CHANGE): veth-osc: link becomes ready
[   18.456744] IPv6: ADDRCONF(NETDEV_CHANGE): br0: link becomes ready
[   18.465730] br0: port 2(veth-tvh-br) entered blocking state
[   18.465741] br0: port 2(veth-tvh-br) entered forwarding state
[   18.465875] IPv6: ADDRCONF(NETDEV_CHANGE): veth-tvh: link becomes ready
[   18.620480] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote
wakeup
[   18.622287] br0: port 3(eth0) entered blocking state
[   18.622300] br0: port 3(eth0) entered forwarding state
[   19.692512] br0: port 3(eth0) entered disabled state
[   20.153028] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa
0xC5E1
[   20.166829] br0: port 3(eth0) entered blocking state
[   20.174763] br0: port 3(eth0) entered forwarding state
[   20.575884] IPv6: ADDRCONF(NETDEV_UP): vb-openhab: link is not ready
[   20.575922] br0: port 4(vb-openhab) entered blocking state
[   20.575931] br0: port 4(vb-openhab) entered disabled state
[   20.576706] device vb-openhab entered promiscuous mode
[   20.576868] br0: port 4(vb-openhab) entered blocking state
[   20.576880] br0: port 4(vb-openhab) entered forwarding state
[   20.700858] br0: port 4(vb-openhab) entered disabled state
[   21.291864] EXT4-fs (dm-1): mounted filesystem with ordered data mode.
Opts: (null)
[   21.404967] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state
recovery directory
[   21.426990] NFSD: starting 90-second grace period (net 80c64dc0)
[   21.536141] IPv6: ADDRCONF(NETDEV_UP): vb-revproxy: link is not ready
[   21.536171] br0: port 5(vb-revproxy) entered blocking state
[   21.536177] br0: port 5(vb-revproxy) entered disabled state
[   21.536780] device vb-revproxy entered promiscuous mode
[   21.536917] br0: port 5(vb-revproxy) entered blocking state
[   21.536926] br0: port 5(vb-revproxy) entered forwarding state
[   21.710696] br0: port 5(vb-revproxy) entered disabled state
[   23.163571] EXT4-fs (dm-5): mounted filesystem with writeback data mode.
Opts: data=writeback
[   23.255947] IPv6: ADDRCONF(NETDEV_UP): vb-tvhead: link is not ready
[   23.262728] br0: port 6(vb-tvhead) entered blocking state
[   23.262768] br0: port 6(vb-tvhead) entered disabled state
[   23.263451] device vb-tvhead entered promiscuous mode
[   23.355693] IPv6: ADDRCONF(NETDEV_UP): vb-work: link is not ready
[   23.365037] br0: port 7(vb-work) entered blocking state
[   23.365047] br0: port 7(vb-work) entered disabled state
[   23.365808] device vb-work entered promiscuous mode
[   23.365957] br0: port 7(vb-work) entered blocking state
[... 248 lines cut ...]
(worker_thread+0x64/0x568)
[  480.392728] [<80136914>] (worker_thread) from [<8013c81c>]
(kthread+0x10c/0x124)
[  480.392746] [<8013c81c>] (kthread) from [<80108148>]
(ret_from_fork+0x14/0x2c)
[  480.392759] Code: e24cb004 e52de004 e8bd4000 e59033f8 (e5130014)
[  480.392768] ---[ end trace 2a38d63a06706320 ]---
[  480.392773] Fixing recursive fault but reboot is needed!
[  480.530484] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  480.640453] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  480.750456] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  480.860471] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  481.007698] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  481.118457] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  481.120448] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  481.226715] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  481.260451] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  481.370454] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  481.990471] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  482.100449] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  482.210453] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)
[  482.306389] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  482.412840] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  482.535394] em28178 #1: writing to i2c device at 0xc8 failed (error=-5)
[  482.720449] em28174 #0: writing to i2c device at 0xd8 failed (error=-5)

popcornmix, 2017-05-15:
@andersthomson are your own builds less stable than rpi-update ones?

andersthomson, 2017-05-15:
Long story. I've had continuous issues with the rpi kernel and the two tv
dvb-t2 dongles I have. Oopses related to the otg driver. Apparently
upstream devs won't touch it in favor of the now mainline dwc2. Reports
elsewhere suggested that the dvwc2 driver has issues with isoc transfers
though (and the tv stuff uses a ton of it I understand). Anyway, I've
compiled a vanilla 4.11, enabled the dwc2 and got the dt sorted (methinks).

The net result is that it boots, tend to have a few hiccups when probing,
finding my two disks/serial cards/two tv cards, but moves on into
operation. The TV stuff fails to come up though. No life from it and a
stream of i2c errors (during mux scanning, I guess). I kind of hope that
I'm missing a module or something, as this happens right after start.

I'll grab a log once I can get the tv grear from the family.

-Anders

andersthomson, 2017-05-15:
Historically no difference. I started out grabbing the kernels etc from the
rasp github tree manually, and they fail in the same way (gentoo here, so
no rpi-update.) In an attenpt at being able to (help with) debuggning the
thing, I've moved to compiling it myself (and there was the odd config I
wanted to change), and eventually move to vanilla.

I've been through two rpi2 about 5 power adapters and got myself two
self-powered usb hub to remove power issues. No difference. USB related
oopses one way or the other...

/Anders

andersthomson, 2017-05-16:
Alright, here's a boot log of vanilla 4.11.1. I've no idea what causes the
i2c errors, but it's consistent on each boot, and I guess timed with
tvheadend starting to do its thing. Tvheadend shows no evidence of the DVB
adapters doing any good.

Ideas?

/Anders

[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.11.1 (root@rpi2.lan) (gcc version 4.9.4
(Gentoo 4.9.4 p1.0, pie-0.6.4) ) #25 SMP Tue May 16 11:11:15 CEST 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7),
cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing
instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 64 MiB at 0x37000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 16 pages/cpu @ef761000 s35840 r8192 d21504
u65536
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.
Total pages: 239936
[    0.000000] Kernel command line: earlyprintk console=ttyAMA0
bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1
dma.dmachans=0x7f35 bcm2709.boardrev=0xa01041 bcm2709.serial=0x9aa48012
bcm2709.u1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288
bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144
bytes)
[    0.000000] Memory: 873404K/966656K available (9216K kernel code, 1204K
rwdata, 3936K rodata, 2048K init, 481K bss, 27716K reserved, 65536K
cma-reserved, 114688K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc0c00000   (10208 kB)
[    0.000000]       .init : 0xc1100000 - 0xc1300000   (2048 kB)
[    0.000000]       .data : 0xc1300000 - 0xc142d220   (1205 kB)
[    0.000000]        .bss : 0xc142f000 - 0xc14a7640   ( 482 kB)
[    0.000000] SLUB: HWalign=128, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000]  RCU restricting CPUs from NR_CPUS=16 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=4
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000021] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps
every 2147483647500ns
[    0.000053] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff,
max_idle_ns: 1911260446275 ns
[    0.000138] bcm2835: system timer (irq = 33)
[    0.002153] arm_arch_timer: WARNING: Invalid trigger for IRQ16, assuming
level low
[    0.002164] arm_arch_timer: WARNING: Please fix your firmware
[    0.002180] arm_arch_timer: WARNING: Invalid trigger for IRQ17, assuming
level low
[    0.002189] arm_arch_timer: WARNING: Please fix your firmware
[    0.002349] arm_arch_timer: Architected cp15 timer(s) running at
19.20MHz (phys).
[    0.002371] clocksource: arch_sys_counter: mask: 0xffffffffffffff
max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[... 483 lines cut ...]
(error=-5)
[   64.322650] em28xx 1-1.3.2:1.0: writing to i2c device at 0xd8 failed
(error=-5)
[   65.282658] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   66.382703] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   68.582764] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   69.682787] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   70.782804] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   71.892862] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   72.992898] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   74.092900] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   74.972948] em28xx 1-1.3.2:1.0: writing to i2c device at 0xd8 failed
(error=-5)
[   74.983764] em28xx 1-1.3.3:1.0: writing to i2c device at 0xc8 failed
(error=-5)
[   75.092996] em28xx 1-1.3.2:1.0: writing to i2c device at 0xd8 failed
(error=-5)

lategoodbye, 2017-05-16:
@andersthomson In order to keep track of this issue i suggested to open a separate issue for dwc2. I'm interested to get as much as possible dwc2 issues fixed, but unfortunately i'm no USB expert.
It would be helpful in narrowing down the issue, if you reduce the affected device to only 1 device per test. So you could make a good / bad list. I also tried tvheadend, but it's too complex for debugging. Aren't there any simpler commands?

andersthomson, 2017-05-16:
This is on vanilla kernel, so an issue on rpi's github would be out of
place, right?

The testing I've done shows that all the USB devices are working find under
dwc2. Well, at least the disks. I've not yet tested the serial devices (one
card reader, one tellstick home automation dev). But no signs of complaints
there.

DVB cli commands are scarce. I've tested scan-dvb from linuxtv-dvb-apps,
but it didn't go anywhere either (same i2c error, no forward progress
shown).

I'll make another scan-dvb run and see if I can correlate i2c error with
the syscall stream.

-A

lategoodbye, 2017-05-16:
Yes, the mailing lists linux-usb or linux-rpi-kernel would be the right place. But i usually didn't expect so much motiviation. Would be nice if you could provide your scan-dvb cli calls.

JamesH65, 2018-06-28:
Is this issue still valid on latest kernel? 

It's been a year since any comments so this issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.