raspberrypi/linux #1650: bitbake failed for kernel 4.7

yongli3, 2016-09-23:
The error message as below. It seems that the commit id is incorrect?
ERROR: linux-raspberrypi-1_4.7.0+gitAUTOINC+d38b45a212-r0 do_fetch: Fetcher failure: Unable to find revision d38b45a21294eaf01aa7568aaeb161d7553477e9 in branch rpi-4.7.y even from upstream
ERROR: linux-raspberrypi-1_4.7.0+gitAUTOINC+d38b45a212-r0 do_fetch: Function failed: Fetcher failure for URL: 'git://github.com/raspberrypi/linux.git;protocol=git;branch=rpi-4.7.y'. Unable to fetch URL from any source.

pelwell, 2016-09-23:
Future rpi- branches (branches that aren't yet the main development branch, currently rpi-4.4.y) are rebased against upstream commits. Once we start making releases from a branch we stop rebasing and start merging.

The commit you refer to still exists but with a different commit ID. You need to tell whoever gave you the hash that bookmarking commits in our future branches is futile because they are regularly rebased.

yongli3, 2016-10-09:
Thanks @pelwell ! issue closed

closed 2016-10-09 by yongli3