raspberrypi/linux #1652: DWC OTG Error on Port Open

bradfol, 2016-09-23:
**Raspberry Pi 3 Model B**
Kernel version: **`4.4.13-v7+`**
Subsystem: **dwc_otg**

Experiencing an issue opening serial ports to connected Atmel devices over USB. The Pi is configured  using the `dwc_otg.speed=1` argument to make USB run at `full-speed`.

We are running a Node.js app and using the [`serialport` module](https://github.com/EmergingTechnologyAdvisors/node-serialport) to communicate with 8 `full-speed` devices connected using powered hubs. There are no other USB devices connected. When opening the ports, an error is thrown:
```
Error: Error: Input/output error, cannot open /dev/ttyACM1
    at Error (native)
```
**NOTE**: Each time the app is run, a different path is identified as the one causing the error (`ttyACM0` through `ttyACM7`).

This also causes `dmesg` output:
```
[  264.905287] INFO:: schedule_periodic: Insufficient periodic bandwidth for periodic transfer.
[  264.905306] ERROR::dwc_otg_hcd_urb_enqueue:544: DWC OTG HCD URB Enqueue failed adding QTD. Error status -1
[  264.905319] cdc_acm 1-1.5.4:1.0: acm_port_activate - usb_submit_urb(ctrl irq) failed
```

Here is the [`lsusb -v` output](https://gist.github.com/bradfol/59b9ec69ed2c23bab91db5f5b23f5459)

P33M, 2016-09-23:
Why are you using `dwc_otg.speed=1`?

bradfol, 2016-09-26:
@P33M We were experiencing data loss issues. We haven't been able to track down the root cause of the data loss, but adding `dwc_otg.speed=1` ended that issue.

P33M, 2016-09-27:
Up to 80% of a frame is reserved for periodic traffic. At full-speed, the driver assumes that any periodic transfer can occur in any frame, thus bandwidth allocation to any endpoint is global. Full-speed is only 12mbit/s line rate therefore the max byte/s for periodic endpoints in totality is 1000/s. The driver is additionally pessimistic in assuming line stuffing and device response times (which is the right thing to do).

This is a hard limit when running in full-speed mode.

P33M, 2017-05-04:
I have spotted that your hubs are single-TT hubs and your Arduinos expose 64-byte interrupt endpoints. See point 4 under Known Issues here:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md

With multi-TT hubs it'll work a lot better at high speed.

closed 2017-05-04 by P33M