raspberrypi/linux #1653: Need help understanding i2c reset handling

notro, 2016-09-24:
I'm working on getting i2c-bcm2835 up to speed so we might shed i2c-bcm2708.
I have implemented Repeated Start (combined) using the TXW interrupt to know when the transfer is active instead of polling the status register.
Here's my i2c-bcm2835 version: https://gist.github.com/notro/faaadb2c551bf3f2b28c5ce95a6994b2

i2c-bcm2708 does a reset between all transfers like this:
```c
static inline void bcm2708_bsc_reset(struct bcm2708_i2c *bi)
{
	bcm2708_wr(bi, BSC_C, 0);
	bcm2708_wr(bi, BSC_S, BSC_S_CLKT | BSC_S_ERR | BSC_S_DONE);
}
```
If I do the same on i2c-bcm2835 then I get many transfer errors.

Next try was mimicking [i2c_actual_read()](https://github.com/raspberrypi/linux/issues/254#issuecomment-15254134) from the firmware:
```c
static void bcm2835_i2c_reset(struct bcm2835_i2c_dev *i2c_dev)
{
	u32 c;

	c = bcm2835_i2c_readl(i2c_dev, BCM2835_I2C_C);

	c &= ~(BCM2835_I2C_C_INTD | BCM2835_I2C_C_INTT | BCM2835_I2C_C_INTR);
	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_C, c);

	c &= ~(BCM2835_I2C_C_ST | BCM2835_I2C_C_READ);
	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_C, c);

	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_S, BCM2835_I2C_S_CLKT |
			   BCM2835_I2C_S_ERR | BCM2835_I2C_S_DONE);

	c |= BCM2835_I2C_C_CLEAR;
	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_C, c);

	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_C, 0);
}
```
This was better, but I still get one or two errors when I write and read a 4k eeprom 500 times.

In the end this is what passes all my tests, just clearing the FIFO:
```c
static void bcm2835_i2c_reset(struct bcm2835_i2c_dev *i2c_dev)
{
	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_C, BCM2835_I2C_C_CLEAR);
	bcm2835_i2c_writel(i2c_dev, BCM2835_I2C_S, BCM2835_I2C_S_CLKT |
			   BCM2835_I2C_S_ERR | BCM2835_I2C_S_DONE);
}
```
This is what the i2c-bcm2835 driver has been doing all along.

So what is the appropriate action on DONE, CLKT and ERR?

And on unexpected TXW, RXR where we have nothing to give the FIFO and must disable the controller?

notro, 2016-09-24:
> In the end this is what passes all my tests, just clearing the FIFO:

Well, I realise that it's not just clearing the FIFO, it's disabling the controller as well. So perhaps this is *the* solution.
But anyways it would be nice to know some more about this.

lategoodbye, 2016-09-27:
@notro Please repeat your questions at the Raspberry Pi forum at the "Bare metal" board. The answers to my questions about TSENS registers were very helpful.

notro, 2016-09-27:
Thanks for the tip Stefan, but I think I'll just settle with using CLEAR and not knowing the details.
This was supposed to be a quick weekend project, but it kind of spiraled out of hand :-)
I've posted the pathces to the [rpi-kernel mailinglist](http://lists.infradead.org/pipermail/linux-rpi-kernel/).

closed 2016-09-27 by notro