raspberrypi/linux #1654: oops in dwc_otg

andersthomson, 2016-09-25:
Hi,

Using a serial adapter, I managed to capture this oops which unfortunately suffer from bad line wraps (captued using a crippled minicom on a router's USB).

I've indications in the past which suggests a hang or similar in the USB host system, so hopefully this can be of some use.

[log3.txt](https://github.com/raspberrypi/linux/files/491665/log3.txt)

lategoodbye, 2016-09-25:
Do you have a defined scenario to reproduce the NULL pointer dereference?

andersthomson, 2016-09-27:
Nope. The pi is on 24/7, serving tvheadend which tend to poll the tuners over USB continuously.  So I suspect there was USB traffic to one of the tuners at the time.

/Anders

lategoodbye, 2016-09-27:
Please provide all the necessary information about this issue. Don't assume that we take them from #1649 .

Most imporant: is this issue even reproducible?

andersthomson, 2016-09-28:
Certainly. Any idea what you need? Are details preferably provided as files of text, btw?

Meanwhile, here are some details:
lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 2: Dev 6, If 0, Class=Vendor Specific Class, Driver=em28xx, 480M
        |__ Port 3: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 7, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 2: Dev 15, If 0, Class=Vendor Specific Class, Driver=usbfs, 12M
            |__ Port 4: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 1: Dev 11, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 4: Dev 13, If 0, Class=Mass Storage, Driver=usb-storage, 480M

Looks funny, but its two USB powered USB hubs (The 10 pot one has three 4 port hubs inside. I hear that's kind of common).

kernel:
Linux version 4.4.15-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #897 SMP Tue Jul 12 18:42:55 BST 2016

This is the first time I see this oops, and I have no idea how to reproduce it. I check the serial console whenever I detect a reboot or hang.

P33M, 2016-10-14:
Dupe of #1681

closed 2016-10-14 by P33M