raspberrypi/linux #1655: Debug Message DEBUG_TIME_EACH_UPDATE fps allways 0 in fbtft_update_display

gdampf, 2016-09-25:
Line 398 and 399 in fbtft-core.c (rpi-4.4.y) seems to be switched.
fps (frames per second) allways 0, because last update time changed before calculation!

popcornmix, 2016-09-26:
Yes, I agree. fps will always be zero.
@ghollingworth okay for me to change?

popcornmix, 2016-09-26:
Sorry - this isn't @ghollingworth 's driver (I was thinking it was from the Pi display driver).
This is actually upstream driver code. It would be best submitted upstream:
https://www.kernel.org/doc/Documentation/SubmittingPatches
and we can pull it in from there.

gdampf, 2016-09-27:
Additional to switching the two lines, the expression in line 395 has to be negated. A leading "!" works well. The complete patch looks like this:

--- linux/drivers/staging/fbtft/fbtft-core.c.orig       2016-09-27 07:25:30.209652437 +0200
+++ linux/drivers/staging/fbtft/fbtft-core.c    2016-09-27 07:23:37.201794540 +0200
@@ -391,10 +391,9 @@ static void fbtft_update_display(struct

        if (unlikely(timeit)) {
                ts_end = ktime_get();
-               if (ktime_to_ns(par->update_time))
+               if (!ktime_to_ns(par->update_time))
                        par->update_time = ts_start;

-               par->update_time = ts_start;
                fps = ktime_us_delta(ts_start, par->update_time);
                fps = fps ? 1000000 / fps : 0;

@@ -406,6 +405,7 @@ static void fbtft_update_display(struct
                        "Display update: %ld kB/s, fps=%ld\n",
                        throughput, fps);
                par->first_update_done = true;
+               par->update_time = ts_start;
        }
 }

closed 2016-09-27 by gdampf

popcornmix, 2016-09-28:
I've sent an email to the author of the commit that appeared to introduce the fps bug. We'll see if he sorts out a fix.

KsenijaS, 2016-09-28:
Hi, I am the author of the patch that introduced this bug, sorry about that. I don't have a raspberry pi to test this patch. Also I wonder if this would to fix the issue:

--- linux/drivers/staging/fbtft/fbtft-core.c.orig       2016-09-27 07:25:30.209652437 +0200
+++ linux/drivers/staging/fbtft/fbtft-core.c    2016-09-27 07:23:37.201794540 +0200
@@ -391,10 +391,9 @@ static void fbtft_update_display(struct

        if (unlikely(timeit)) {
                ts_end = ktime_get();
-               if (ktime_to_ns(par->update_time))
+               if (!ktime_to_ns(par->update_time))
                        par->update_time = ts_start;

-               par->update_time = ts_start;
                fps = ktime_us_delta(ts_start, par->update_time);
+              par->update_time = ts_start;
                fps = fps ? 1000000 / fps : 0;

@@ -406,6 +405,7 @@ static void fbtft_update_display(struct
                        "Display update: %ld kB/s, fps=%ld\n",
                        throughput, fps);
                par->first_update_done = true;
        }
 }

There is a more simple tutorial how to send a patch: https://kernelnewbies.org/FirstKernelPatch
I can send @gdampf patch, but I would also like to give him/her credit for it

KsenijaS, 2016-09-28:
Hi, I am the author of the patch that introduced this bug, sorry about that. I cannot test this patch since I don't have the hardware. I also wonder if this would work:
```
--- a/drivers/staging/fbtft/fbtft-core.c
+++ b/drivers/staging/fbtft/fbtft-core.c
@@ -391,11 +391,11 @@ static void fbtft_update_display(struct fbtft_par *par, un
 
        if (unlikely(timeit)) {
                ts_end = ktime_get();
-               if (ktime_to_ns(par->update_time))
+               if (!ktime_to_ns(par->update_time))
                        par->update_time = ts_start;
 
-               par->update_time = ts_start;
                fps = ktime_us_delta(ts_start, par->update_time);
+               par->update_time = ts_start;
                fps = fps ? 1000000 / fps : 0;
 
                throughput = ktime_us_delta(ts_end, ts_start);
```

There is a more simple tutorial how to send a patch: https://kernelnewbies.org/FirstKernelPatch
I can send @gdampf patch but I would also like to give him/her credit for it.