raspberrypi/linux #1656: 4.4.22-1 causes `lsmod` to return empty and my wlan0 device to disappear from ifconfig

ric2b, 2016-09-26:
I'm running Arch Arm on a Raspberry Pi A+ but running the OS on a model B yelds the same result. 
After updating to 4.4.22-1 my wlan0 device disappeared from `ifconfig -a` and calling `lsmod` yelded 0 results.
The usb wireless card is a TL-WN725N V2, it uses the r8188eu driver module and was working perfectly on 4.4.21-1 (after disabling the power saving features that caused it to shutdown after a while).
I downgraded the package to 4.4.21-1 and both issues were fixed. Updated again to 4.4.22-1 just in case it was an unrelated issue and the issues came back as soon as I rebooted. Downgraded again and am currently running 4.4.21-1 without issues (over 10h uptime without losing the connection).

Every other distro package is updated and I don't have many extra packages installed (sudo, mlocate, vim, deluge, rtorrent, fail2ban and I can't recall if there's anything else)

I apologise if I'm not in the right place to report the issue.

popcornmix, 2016-09-26:
Not seeing any commits related to rtl8188eu here: https://cdn.kernel.org/pub/linux/kernel/v4.x/ChangeLog-4.4.22
I'm afraid we can't offer support for Arch. Can you confirm if using a raspbian jessie image with rpi-update (which will get you 4.4.22) has the same issue?
Does updating to newest 4.4.21 kernel (``sudo rpi-update 21505f3fb5d8e2a5abbf95baf441a7a6415d4a82``) doesn't have the issue?

pelwell, 2016-09-26:
You could run lsusb and dmesg with both kernels and look for differences (or post the results here so we can look).

ric2b, 2016-09-27:
I don't think I'll have time to test Raspbian in the next few days, maybe over the weekend, sorry.

Ruffio, 2017-02-05:
@ric2b how did the test go? Can this issue be closed?

ric2b, 2017-02-05:
Yes, you can close it, sorry. I ended up installing Raspbian and forgot about this, it's running perfectly :)

closed 2017-02-05 by popcornmix