raspberrypi/linux #1659: How to deal with that "/dev/root" is full?

zfyseu, 2016-09-29:
Hello,everyone!
I have installed nginx,apparmor on my raspberry pi based on raspbian.

However when I reboot, the desktop can't start.
And when run the command "df -h" through putty, it return with the following:
[image]

Is there anyone know how to deal with it ?
Thanks!

vk5tu, 2016-09-29:
Hi @zfyseu. This community is for kernel bugs related to the Raspberry Pi. You might want to try the forums at [www.raspberrypi.org/forums](https://www.raspberrypi.org/forums/) for advice on solving issues such as this one.

To get you started, the likely issues are either
* the 4GB SD card you are using has been filled. Not too surprising, 4GB is too small.
* the filesystem wasn't expanded to use the remainder of the SD card when booting for the first time. In recent Raspbian this should have happened automatically. In any case, see the option in `sudo raspi-config`. You'll likely have to re-install, since many files will have been lost due to the full filesystem. Alternatively see `debsums` to verify the installation.

(By the way, when asking for help on GitHub it is best to copy screen output using cut-and-paste. Then mark the block as monospaced -- GitHub uses ``` for this. Graphical screen dumps of text are unwieldy and make it harder to answer your question well).

zfyseu, 2016-09-29:
Thanks!

来自 魅族 MX4

时间：周四 9月29日 11:42
收件人：raspberrypi/linux <linux@noreply.github.com>
抄送：zfyseu <zfyseu@163.com>,Mention <mention@noreply.github.com>

zfyseu, 2016-09-29:
Thanks,vk5tu!
I have solved the problem according to your advice.
I have expanded the filesystem by raspi-config command.

closed 2016-09-29 by zfyseu