raspberrypi/linux #1660: System freeze when make a bridge to the onboard ethernet

sincoder, 2016-09-29:
I setup a hostapd on a wifi interface : wlan2 ，and bridge it to the onboard ethernet : eth0

brctl addbr br0
brctl addif br0 eth0
brctl addif br0 wlan2

then the access point can connect to internet ,but after a while ，system will lose response  
then  i  use a  usb ethernet connect to the  pi , i got :  eth1 。when  bridge to eth1，all works fine 。
i also test the onboard wifi :  wlan0 , have the same issue .

sincoder, 2016-09-29:
my pi is :  pi 3 mode B
system is updated :  apt update   apt upgrade rpi-update

LoyalNine1487, 2016-10-20:
1. Have you recompiled the kernel your self?
2. Can you paste you logs so we can see whats going wrong?
3. How many interfaces are you trying to bridge (4?)

JamesH65, 2017-05-04:
I believe this is now fixed. Please retest with the latest Raspbian, and close if now OK.

sincoder, 2017-05-08:
@JamesH65 when i get my new raspi , i will test it

sincoder, 2017-05-08:
@JamesH65 i have test it, all my wireless card just can not make a bridge. brctl addbr br0 wlan* always fail. is this the way your fix the bug ? your just block this issue. i do not think bridge to a wlan interface is useless.

pelwell, 2017-05-08:
> is this the way your fix the bug ? your just block this issue.

@JamesH65 has been very helpful so far, fixing bugs in multiple drivers and solving bridging problems for many people. If it isn't working for you then there must be something different about your configuration - either hardware or software - that is either responsible for the failure or causing another bug to show itself. We aren't short of users with problems, and most of them have better attitudes than you, so try not to take your frustration out on people who are trying to help.

JamesH65, 2017-05-08:
The bug fix recently released fixed a coding fault in the wireless and ethernet drivers when used in bridging mode that mean that the ethernet would fail after a random time (could be days). 

Sounds like your fault might be different. If you follow the known working instructions here on the latest Raspbian (make sure you are up to date) https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md does the bridging work?

Do you have any message logs when it fails?

sincoder, 2017-05-08:
@JamesH65 OK, i will do a further test. i do not see any error log use dmesg

sincoder, 2017-05-08:
@pelwell  sorry . i misunderstand the fix is just make bridge to wlan interface do not work .

sincoder, 2017-05-08:
@JamesH65 retest it , have all the packages up to date, run rpi-update and reboot.  then nothing change 
root@raspberrypi:~# killall wpa_supplicant 
wpa_supplicant: no process found
root@raspberrypi:~#  brctl addif br0  wlan0
can't add wlan0 to bridge br0: Operation not supported
i have test the onboard wlan  interface and a extra  wireless card . alaway told me "Operation not supported" 
bridge eth0 is ok
test on my old pi2 ,
root@pi:~# brctl addbr br0
root@pi:~# brctl addif br0 wlan0
 root@pi:~# uname -a
Linux pi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

report no problem

JamesH65, 2017-05-08:
rpi-update is not really advised as it may bring in a non-working/testing kernel. Just apt-get update/upgrade should be sufficient. 

Does the same occur on a new raspbian install? You appear to be logged in as root so clearly it's not standard. A quick google seems to provide a number of different reasons why this error may occur, without knowing more about your setup its difficult to know which it may be.

JamesH65, 2017-05-08:
Actually, looking at your commands, you should be doing something like this...

sudo brctl addbr br0
sudo brctl addif br0 eth0 wlan0   <---- your command has insufficient parameters. You want br0 to bridge between two specific ports.

sincoder, 2017-05-08:
@JamesH65 i am do a new install now. download the image from official site and reinstall in a new pi3 board, then the same thing happed. 
pi@raspberrypi:~ $ sudo brctl show
bridge name	bridge id		STP enabled	interfaces
br0		8000.000000000000	no		
pi@raspberrypi:~ $ sudo brctl addif br0 eth0
pi@raspberrypi:~ $ sudo brctl addif br0 wlan0
can't add wlan0 to bridge br0: Operation not supported
pi@raspberrypi:~ $ 

above command are the first command i type on this pi3 board and have not change any config file . newest install with all default config

sincoder, 2017-05-08:
@JamesH65 i found the problem. the wlan interface must be put into master mode ,then it become bridgeable .

>  pi@raspberrypi:~ $ **iwconfig wlan0**
> wlan0     IEEE 802.11bgn  ESSID:off/any  
>           Mode:Managed  Access Point: Not-Associated   
>           Retry short limit:7   RTS thr:off   Fragment thr:off
>           Power Management:off
> pi@raspberrypi:~ $ **sudo brctl addif br0 wlan0**
> can't add wlan0 to bridge br0: Operation not supported
> pi@raspberrypi:~ $ **iwconfig wlan0**
> wlan0     IEEE 802.11bgn  Mode:Master  
>           Retry short limit:7   RTS thr:off   Fragment thr:off
>           Power Management:off
>           
> pi@raspberrypi:~ $ **sudo brctl addif br0 wlan0**  <--  no error

I didn't  notice this before .now i will create a bridge and test how stable it is

JamesH65, 2017-05-08:
How did it get set to Managed Access Point in the first place - was this something you  did? Because by default it should be in master mode I think.

sincoder, 2017-05-08:
@JamesH65 No, I reinstall  it and do not do any config even connect to a wifi hotspot. and the Manager mode is the default mode. until i run hostapd on the interface , it turn to Master mode.

sincoder, 2017-05-08:
test it for 5 hours , it works prefect !

sincoder, 2017-05-08:
@JamesH65 thanks for your fix ! a great work!

closed 2017-05-08 by sincoder