raspberrypi/linux #1661: Problems with rfid-rc522 and the Raspberry Pi 3 Model B v1.2

sven-hildebrandt, 2016-09-30:
We are trying to scan RFID-Chips with rfid-rc522 and the Raspberry Pi 3 Model B v1.2, but it does not work for us.

We did it exactly like described [here](https://github.com/raspberrypi/linux/issues/1547#issuecomment-245220114).

```
dtoverlay -l
Overlays (in load order):
0:  dtparam  spi=on

dmesg | grep -i spi
No ouput

dmesg > ~/dmesg.txt

lsb_release -a
No LSB modules are available.
Distributor ID:Raspbian
Description:Raspbian GNU/Linux 8.0 (jessie)
Release:8.0
Codename:jessie

vim /boot/config.txt
# Its default. Just "dtparam=audio=on” is enabled.

~/SPI-Py $ sudo python setup.py install
running install
running build
running build_ext
running install_lib
running install_egg_info
Removing /usr/local/lib/python2.7/dist-packages/SPI_Py-1.0.egg-info
Writing /usr/local/lib/python2.7/dist-packages/SPI_Py-1.0.egg-info

~/MFRC522-python $ sudo python Read.py
Welcome to the MFRC522 data read example
Press Ctrl-C to stop
```
[dmesg.txt](https://github.com/raspberrypi/linux/files/502568/dmesg.txt)

**Nothing happens.. We have tested several different RFID chips..**

PS: We have added "dtoverlay=spi0-hw-cs” to /boot/config.txt
```
gpio readall
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5V      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 0 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 1 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 1 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | ALT0 | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | ALT0 | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 1 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+——

dmesg | grep -i spi
No ouput
```

Its still not working :-(

The Wire:
[image]
[image]
[image]
[image]

---

Answer from: @pelwell 
> I'm happy to help (although I'd hoped this matter was closed), but I'd
> rather do it in public where other people can benefit from the exchange.
> Please add to the github issue, or open a thread on the Forum (or find
> an existing one).
> 
> Note that the SPI driver is quiet, so searching dmesg won't help. If you
> want to check the SPI driver is loaded and functional, "ls /dev/spi*".
> 
> And in case it wasn't clear, the dtoverlay is only used up to the next
> reboot. After a reboot the firmware uses the dtparam in config.txt to
> enable the SPI interface in the kernel Device Tree.
> 
> I will need to know the version of the SPI-Py library in use - run "git
> branch -v" in the SPI-Py directory.
> 
> Phil

---

hi @pelwell, thank you very much for your answer! does this help?
```
ls -1 /dev/spi*
/dev/spidev0.0
/dev/spidev0.1

~/SPI-Py $ git branch -v
* master 69183da Merge pull request #18 from pelwell/master
  ßv     69183da Merge pull request #18 from pelwell/master

~/MFRC522-python $ git branch -v
* master 8ddc994 Read example now loops until Ctrl-C is caught.
```

pelwell, 2016-09-30:
I have recreated your setup (I have exactly the same module) by ignoring my instructions and following your wiring, and it works for me.

A few things to note:

1) spi-hw-cs is no longer necessary (the fixed SPI-Py library takes care of that), but it does no harm in this case.

2) If you've used raspi-config to enable the SPI interface then your config.txt should also include "dtparam=spi=on", but running "sudo dtparam spi=on" from the console has the same effect. Either way, provided /dev/spidev0.0 exists then SPI has been enabled correctly (provided you aren't using a dedicated SPI device driver that claims CE0/CS0).

3) I had a local modification to MFRC522.py that modifies line 110 to initially set the NRSTPD to 0 (was 1) to ensure that the device is reset, but it also works without that change.

You don't mention your power supply - make sure it is

sven-hildebrandt, 2016-09-30:
we have did 2) and 3), but its still not working :-(

> You don't mention your power supply - make sure it is 
The power comes over usb (to the raspberry). And the usb is connected to a power socket (and not to a pc). Is there more we have to be aware of?

pelwell, 2016-09-30:
My comment was submitted prematurely - it now says:

> You don't mention your power supply - make sure it is at least capable of delivering 2.5A.

You can see if power has been an issue by running `vcgencmd get_throttled`. If it returns anything other than `throttled=0x0` then either voltage or temperature has been an issue.

Do you have another RFID module to try? If you really want to understand what is going wrong I can recommend hooking a logic analyser up to the pins (I have a Saleae Logic Pro16 which includes an SPI protocol decoder) and looking for activity.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-18 by JamesH65

eDigirolamo, 2019-11-06:
I have a similar issue, It will only read cards after starting the file twice

pelwell, 2019-11-06:
If you follow my guide (#2613) it should work.