raspberrypi/linux #1662: Time does not Sync (auto-update) over Wifi on Raspberry Pi 3

david-a-wheeler, 2016-10-01:
Time/date does not sync at boot when using my Pi3 over Wifi (a really common situation).  This is a real problem - for example, many HTTPS websites won't allow connections with such bad date/time settings.

A workaround is described here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=141454

In short, adding this to /etc/rc.local fixes it:
/sbin/iptables -t mangle -I POSTROUTING 1 -o wlan0 -p udp --dport 123 -j TOS --set-tos 0x00

That's a hackish workaround; this needs a real fix.  This seems to suggest some firmware/driver defect in RPI3. It seems unable to forward IP packets with ToS != 0, and  ntpd emits IP/UDP packets with ToS field set to 0xc0.

Not sure exactly where the problem is... but please fix!!!

clivem, 2016-10-01:
I believe this is a duplicate of #1519

david-a-wheeler, 2016-10-01:
Yes, it's a dup. Thanks. Odd, I could not find that earlier.  I hope a real fix can be found soon.  I suspect this is a really common problem, but many people don't know how to report it. Good luck.

closed 2016-10-01 by david-a-wheeler