raspberrypi/linux #1663: in DVI modes, audio output dosen't work ,and in HDMI modes ,audio work,but the resolution is too bad.

superbool, 2016-10-02:
I install the image `RASPBIAN JESSIE WITH PIXEL` latest,and use the 14' LED monitor with horn .
I use HDMI connect the pi 3 and the monitor.and it looks work well.
But defalut in DVI modes, audio output dosen't work ,when I change the `/boot/config.txt` hdmi_drive=2` audio works,but the resolution is too bad. How can I solve it.

popcornmix, 2016-10-03:
DVI doesn't support audio, so that behaviour is expected.
What is the problem with the resolution?
What do ``tvservice -s`` and ``tvservice -m CEA`` and ``tvservice -m DMT`` report?
What resolution do you believe the monitor to be natively?

superbool, 2016-10-03:
Look at the two pictures of the screen,they ware toked by my phone:

picture1  HDMI Mode: 

[image]
picture2 DVI Mode: 

[image]

Obviously,picture2 is more clear than picture1.The text in picture1 has more aliasing.

This is the report:
```
HDMI Mode
pi@raspberrypi:~ $ tvservice -s
state 0x12000a [HDMI DMT (39) RGB full 16:9], 1360x768 @ 60.00Hz, progressive
pi@raspberrypi:~ $ tvservice -m CEA
Group CEA has 16 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
           mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced
           mode 6: 720x480 @ 60Hz 4:3, clock:27MHz x2 interlaced
           mode 7: 720x480 @ 60Hz 16:9, clock:27MHz x2 interlaced
  (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive
           mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced
           mode 21: 720x576 @ 50Hz 4:3, clock:27MHz x2 interlaced
           mode 22: 720x576 @ 50Hz 16:9, clock:27MHz x2 interlaced
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive
           mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive
pi@raspberrypi:~ $ tvservice -m DMT
Group DMT has 6 modes:
           mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 8: 800x600 @ 56Hz 4:3, clock:36MHz progressive
           mode 9: 800x600 @ 60Hz 4:3, clock:40MHz progressive
           mode 16: 1024x768 @ 60Hz 4:3, clock:65MHz progressive
           mode 22: 1280x768 @ 60Hz 15:9, clock:68MHz progressive
  (prefer) mode 39: 1360x768 @ 60Hz 16:9, clock:85MHz progressive

DVI Mode
pi@raspberrypi:~ $ tvservice -s
state 0x120006 [DVI DMT (39) RGB full 16:9], 1360x768 @ 60.00Hz, progressive
pi@raspberrypi:~ $ tvservice -m CEA
Group CEA has 16 modes:
           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive
           mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive
           mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive
           mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced
           mode 6: 720x480 @ 60Hz 4:3, clock:27MHz x2 interlaced
           mode 7: 720x480 @ 60Hz 16:9, clock:27MHz x2 interlaced
  (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive
           mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive
           mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive
           mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive
           mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced
           mode 21: 720x576 @ 50Hz 4:3, clock:27MHz x2 interlaced
           mode 22: 720x576 @ 50Hz 16:9, clock:27MHz x2 interlaced
           mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive
           mode 32: 1920x1080 @ 24Hz 16:9, clock:74MHz progressive
pi@raspberrypi:~ $ tvservice -m DMT
Group DMT has 6 modes:
           mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive
           mode 8: 800x600 @ 56Hz 4:3, clock:36MHz progressive
           mode 9: 800x600 @ 60Hz 4:3, clock:40MHz progressive
           mode 16: 1024x768 @ 60Hz 4:3, clock:65MHz progressive
           mode 22: 1280x768 @ 60Hz 15:9, clock:68MHz progressive
  (prefer) mode 39: 1360x768 @ 60Hz 16:9, clock:85MHz progressive
```

But I also have got two screen shot pictures in two modes:

picture3 HDMI Mode:  

[image]
picture4 DVI Mode:  

[image]

They are almost the same，there is no different. But in the visible they have some differents.

Maybe the monitor's problem. I'll test it on an another monitor.

Thank you very much.

closed 2017-05-18 by JamesH65