raspberrypi/linux #1670: Serial number in Device Tree

notro, 2016-10-08:
Currently the serial number is passed in 2 ways by the bootloader:
- Kernel commandline as text: bcm270{8,9}.serial= (used by bcm2708/9)
- Device Tree in binary form: /system/linux,serial (used by downstream bcm2835)

This is the only in-kernel user of this information:
```
$ cat /proc/cpuinfo | grep Serial
Serial          : 00000000316e5228
```

Linux 4.2 introduced a way to pass in the serial number via Device Tree as a string:
Commit: https://github.com/torvalds/linux/commit/3f599875e5202986b350618a617527ab441bf206
Doc: https://github.com/torvalds/linux/commit/13dd92bb4599b5655cafe1f2c0365396a096b94a

```
/ {
    serial-number = "00000000316e5228";
};
```

@pelwell can we have it a third way? :-)

pelwell, 2016-10-10:
Sure - I've dropped a patch in the pipe.

popcornmix, 2016-10-11:
This is in latest rpi-update firmware.

notro, 2016-10-11:
Thanks, that worked just fine.

closed 2016-10-11 by notro

kukabu, 2016-10-12:
@popcornmix what about implement local-mac-address in DT? We can use the https://github.com/raspberrypi/linux/commit/c489565b536ff5382460273fd9513f0adebec024 and drop the https://github.com/raspberrypi/linux/commit/1bc556871ead5d8c9f1c2721523d5b80365d308f

kukabu, 2016-10-12:
WOW! The mac-address exists in DT, it seems we can drop the https://github.com/raspberrypi/linux/commit/1bc556871ead5d8c9f1c2721523d5b80365d308f.
But can you rename the mac-address to local-mac-address? The mac-address is better use for user-defined whereas local-mac-address - vendor-defined.

pelwell, 2016-10-12:
The of_get_mac_address() function tries "mac-address" first, then "local-mac-address", then "address", so using "mac-address" is actually a performance optimisation.

But seriously, can you think of a situation where defining "local-mac-address" instead is of some practical benefit?

kukabu, 2016-10-12:
doc says:
- local-mac-address: array of 6 bytes, specifies the MAC address that was
  assigned to the network device;
- mac-address: array of 6 bytes, specifies the MAC address that was last used by
  the boot program; should be used in cases where the MAC address assigned to
  the device by the boot program is different from the "local-mac-address"
  property;

I think a one 'if' doesn't significantly degrade performance but the user may be able to change the mac address via DT.

pelwell, 2016-10-12:
I don't see how it is particularly useful retaining the original MAC address (except perhaps for information), but I agree that our usage doesn't really match the spec.

Create a new issue for tracking purposes (one problem per issue, please) and I'll change it (assuming there are no problems).