raspberrypi/linux #1671: rpi-4.8.y build failure

asavah, 2016-10-09:
rpi04.8.y branch fails to build with (cross)gcc-6.2
```
drivers/media/platform/bcm2835/bcm2835-camera.c:656:17: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .queue_setup = queue_setup,
                 ^~~~~~~~~~~
drivers/media/platform/bcm2835/bcm2835-camera.c:656:17: note: (near initialization for ‘bm2835_mmal_video_qops.queue_setup’)
```

popcornmix, 2016-10-09:
Try grabbing https://github.com/raspberrypi/linux/pull/1667/commits/6573f071ef1f347a83e0853df95e7837fabdfe44

asavah, 2016-10-09:
Thx, will rebuild in a few mins, from the looks if it - this should work,
with my limited kernel knowledge I've already traced the upstream change to https://github.com/raspberrypi/linux/commit/36c0f8b32c4bd4f668cedfba6d97afaa84f055fb

Will report back asap.

asavah, 2016-10-09:
@popcornmix The suggested fix works like a charm.
Clean build with defconfig. Thx.
Feel free to close once/if you merge the fix.

asavah, 2016-10-09:
Fixed in https://github.com/raspberrypi/linux/commit/88403aeb5009ffa3cf172bfaa06087ee2ee3cabe

closed 2016-10-09 by asavah