raspberrypi/linux #1672: Proto Audio Card (WM8731) wrong sampling clock

bg3mdo, 2016-10-09:
I am currently working on a project, which porting a WM8731 (proto audio card) to Pi 3.

HW and SW are based on: https://github.com/raspberrypi/linux/issues/1302

'rpi-proto audio codec on Raspberry Pi v2 model B #1302'

Now I am encountering some problems, where my LRCK (sample rate) is much higher than I expect. I can get sound played, but the sound is pitched. 

Here is my father testing method:

CODE: SELECT ALL
speaker-test -Dplughw:sndrpiproto -f 400 -F S16_LE -r 48000 -c 2 -t sine

I was using scope to check the LRCK rate under different sample rates, what I got as follows:

96K what I got 125K on Pi
48K ---> 62.5K
32K ---> 41.68K
8K ---> 10.42K

Cheers.

bg3mdo, 2016-10-09:
Tried different device trees, where I tried audio_injector sound card one, they have the same chipset.  I have the same problem -- sampling clock is wrong.

bg3mdo, 2016-10-09:
Seems previous measure is not very accurate 

96K what I got 128k on Pi
48K ---> 64K
32K ---> 42.68K

bg3mdo, 2016-10-09:
Here something might be interesting ... 1.2G (Pi 3) CPU rate vs 900MHZ (Pi 2)
128k/96k = 1.2G/900mhz....   seems no one report problem if they are using Pi2

pelwell, 2016-10-09:
The I2S clock is independent of the ARM and VPU (core) clocks, so although it is a nice theory I don't think the Pi2/Pi3 clock speeds are relevant.

Are you building your own kernel? If so, could you try adding PR #1666 to see if it has an effect?

bg3mdo, 2016-10-10:
RS component (UK) seems sending Audio Codec Proto with a wrong crystal. The WM8731 should work as a master device, which sending clocks to Pi.  RS component one has 16MHz crystal other than 12.288Mhz. This is what I get, for example, 64k when setting as 48k   ---- 64k/48k = 16/12m...

closed 2016-10-10 by bg3mdo