raspberrypi/linux #1674: VC4 doesn't work in newer kernels

Gerharddc, 2016-10-10:
I am trying to get VC4 running on the RPi3 with the 4.8 kernel but am getting the following errors:

`root@raspberrypi3:~# dmesg | grep vc4                                                                
[    1.376443] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517                
[    1.376529] vc4-drm soc:gpu: master bind failed: -517                                             
[    4.101281] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517                
[    4.109058] vc4-drm soc:gpu: master bind failed: -517                                             
[    4.892690] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517                
[    4.892775] vc4-drm soc:gpu: master bind failed: -517                                             
[    4.914969] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops)                               
[    4.915156] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops)                         
[    4.915292] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops)                         
[    4.915423] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops)                         
[    4.915454] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops)                                 
[    4.915501] [drm:vc4_v3d_bind] *ERROR* V3D_IDENT0 read 0xdeadbeef instead of 0x02443356           
[    4.915530] vc4-drm soc:gpu: failed to bind 3fc00000.v3d (ops vc4_v3d_ops): -22                   
[    4.915769] vc4-drm soc:gpu: master bind failed: -22                                              
[    4.915798] vc4-drm: probe of soc:gpu failed with error -22 `

I get the same result on 4.7 and haven't tried another one. This is not a problem on 4.4. I also tried using the very latest firmware like start.elf from the firmware repository but it made no difference.

popcornmix, 2016-10-11:
vc4-kms-v3d is only supported on 4.4 kernel currently.
The patches come from https://github.com/anholt/linux so you could try requesting downstream 4.8 support from there.

closed 2016-10-11 by popcornmix

popcornmix, 2016-10-31:
FYI, the 4.9 tree is closer to upstream, and I was able to get the ``vc4-kms-v3d`` and ``vc4-fkms-v3d``overlays working with it.
The main issue with the "0xdeadbeef" error in log was the missing power-domain for v3d
https://github.com/raspberrypi/linux/commit/68c80cd8059d483455fabfc8ac3a6de6500b9756#diff-feed3f61cccdcb4bdf658b6918a83380R144

agherzan, 2016-10-31:
@popcornmix I see that 4.8 doesn't have vc4-fkms-v3d . Any idea why? Maybe it's just a bug.

popcornmix, 2016-10-31:
Feel free to cherry-pick commits and we'll accept a PR if they work.
But the experimental graphics driver is not something we directly maintain.
@anholt provides patches for 4.4 kernel so that is expected to work.
As 4.9 seems to be working and most of the code is upstream in that tree, then newer kernels are likely to work, but older ones may need the right set of commits to be cherry picked.

agherzan, 2016-10-31:
Thanks @popcornmix for the clarification.