raspberrypi/linux #1676: Yocto: Compile error, kernel version 4.4.23

latiaoxia, 2016-10-11:
log:

```
DEBUG: Executing shell function do_compile
NOTE: make -j 2 zImage CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd LD=arm-poky-linux-gnueabi-ld.bfd  
  GEN     ./Makefile
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  GEN     ./Makefile
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/seccomp.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/dtc/dtc.o
  Using /home/max/project/yocto-build/tmp/work-shared/myrpi2/kernel-source as source for kernel
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  SHIPPED scripts/genksyms/parse.tab.c
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
[... 147 lines cut ...]
 #define FIXMAP_PAGE_NORMAL (FIXMAP_PAGE_COMMON | L_PTE_MT_WRITEBACK)
                                                  ^
/home/max/project/yocto-build/tmp/work-shared/myrpi2/kernel-source/include/asm-generic/fixmap.h:64:26: note: in expansion of macro 'FIXMAP_PAGE_NORMAL'
  __set_fixmap(idx, phys, FIXMAP_PAGE_NORMAL)
                          ^
/home/max/project/yocto-build/tmp/work-shared/myrpi2/kernel-source/arch/arm/kernel/patch.c:39:2: note: in expansion of macro 'set_fixmap'
  set_fixmap(fixmap, page_to_phys(page));
  ^
/home/max/project/yocto-build/tmp/work-shared/myrpi2/kernel-source/scripts/Makefile.build:258: recipe for target 'arch/arm/kernel/patch.o' failed
make[3]: *** [arch/arm/kernel/patch.o] Error 1
/home/max/project/yocto-build/tmp/work-shared/myrpi2/kernel-source/Makefile:959: recipe for target 'arch/arm/kernel' failed
make[2]: *** [arch/arm/kernel] Error 2
make[2]: *** Waiting for unfinished jobs....
  AS      arch/arm/mm/abort-ev7.o
  AS      arch/arm/mm/pabort-v7.o
  AS      arch/arm/mm/cache-v7.o
  AS      arch/arm/mm/proc-v7.o
  LD      arch/arm/mm/built-in.o
Makefile:150: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make: *** [__sub-make] Error 2
ERROR: oe_runmake failed
ERROR: Function failed: do_compile (log file is located at /home/max/project/yocto-build/tmp/work/myrpi2-poky-linux-gnueabi/linux-rpi2/4.4.23+gitAUTOINC+844d3dfedc-r0/temp/log.do_compile.23599)
```

latiaoxia, 2016-10-11:
I use the bacm2709_defconfig.

asavah, 2016-10-11:
Compare .config generated from _defconfig by yocto and simple cross-compiling (as seen in your SO question).
My guess would be that somehow .config generated by yocto does not have CONFIG_MMU=y .
This something that should be addresed by yocto developers and their weird toolchains.

popcornmix, 2016-10-11:
Are you setting ARCH=arm when running make?

latiaoxia, 2016-10-12:
BTW, if I cross compile manually ,everything is ok, so the problem comes from yocto, not this linux kernel.

closed 2016-10-12 by latiaoxia

wigcheng, 2016-11-02:
I have the same issue, how to fix? modify the do_compile script?

skrawn, 2017-02-01:
I fixed this by adding my own do_configure function:

`do_configure() {
    oe_runmake -C ${S} O=${B} ${KBUILD_DEFCONFIG}
}`

Works as long as you have KBUILD_DEFCONFIG somewhere.