raspberrypi/linux #1679: Mac addess in Device Tree

kukabu, 2016-10-12:
Currently the firmware sets mac-address node. It doesn't meet specifications.
doc says:
- local-mac-address: array of 6 bytes, specifies the MAC address that was
  assigned to the network device;
- mac-address: array of 6 bytes, specifies the MAC address that was last used by
  the boot program; should be used in cases where the MAC address assigned to
  the device by the boot program is different from the "local-mac-address"
  property;

It's useful with https://github.com/raspberrypi/linux/commit/c489565b536ff5382460273fd9513f0adebec024 for dropping https://github.com/raspberrypi/linux/commit/1bc556871ead5d8c9f1c2721523d5b80365d308f

pelwell, 2016-10-12:
I'll make the change in the firmware, and I can guarantee it won't break anything because the smsc95xx driver isn't using it - as a USB device it doesn't have an associated Device Tree node (dev->udev->dev.of_node == NULL), so of_get_mac_address() will always fail.

kukabu, 2017-03-01:
Fixed.

closed 2017-03-01 by kukabu