raspberrypi/linux #1680: clock_gettime() granularity loss?

garyemiller, 2016-10-12:
I run NTPsec on a number of Raspberry Pi's.  They all run on the same kernel config, with NO_HZ=off.

When ntpd starts, it runs a tight loop on clock_gettime() to find the smallest time interval possible between calls.  On 4.4.22, and before, that answer was always just over 1 micro Second.  In 4.4.24 it is now double that.  PPS timing is accurate to better than 1 micro Second, so this loss becomes obvious to NTP implementations.

Was this intentional?  Is there a workaround?

garyemiller, 2016-10-12:
This issue is not present on Intel hosts.

popcornmix, 2016-10-13:
I'm not aware of any change that would cause this, so it's not intentional from here.
Probably an upstream commit.
If you are building your own kernel then bisecting should be possible to identify the commit responsible.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.
Please update to the latest kernel which may contain a fix for this issue.

closed 2017-05-18 by JamesH65