raspberrypi/linux #1683: 4.4.23-v7+ console redirect

gitbf, 2016-10-14:
I've been using cmdline.txt with a "console=tty3" entry to suppress console (HDMI video) output during boot (used in combination with an init.d **fbi** script to display a custom splash image). Now however, scrolling boot messages are displayed irrespective of this cmdline change.

Is this intentional, is there a workaround?

pelwell, 2016-10-14:
How have you updated to 4.4.23? Which was the last kernel version that worked as you want it to?

gitbf, 2016-10-14:
Base is Raspbian Jessie Lite (May 2016)

apt-get update
apt-get dist-upgrade
rpi-update

No record of last working kernel (my guess is the change came with the most recent full release, 2016-9-23).

Ruffio, 2017-02-05:
@gitbf had this been resolved?

gitbf, 2017-02-05:
No - console redirection is no longer a viable method towards a clean boot (now using kernel 4.4.38). This in itself is not so much of an issue for me, but some way to configure for a clean boot (no console boot messages displayed whatsoever) is desired.

Disabling console login and adding cmdline options "quiet splash vt.global_cursor_default=0 loglevel=3" is close as this will remove everything except output from "systemd-fsck". 

[image].

Can these messages be suppressed as well?

popcornmix, 2017-02-06:
> Can these messages be suppressed as well?

Not by the kernel. This is being added after kernel has booted.

May be some ideas here:
https://wiki.archlinux.org/index.php/Silent_boot

closed 2017-05-18 by JamesH65