raspberrypi/linux #1690: usb_modeswitch udev-rules doesn't work in Rasbian Lite

mcgurk, 2016-10-19:
(Raspbian Jessie Lite, Version:September 2016, Release date:2016-09-23, Kernel version:4.4)

Huawei 3G modem doesn't automatically trigger succesfully usb_modeswitch in Raspbian Lite.

I tracked down problem to udev-rule parameters.
%b is interpreted differently in /lib/udev/rules.d/40-usb_modeswitch.rules.
Line:
ATTRS{idVendor}=="12d1", ATTR{bInterfaceNumber}=="00", ATTR{bInterfaceClass}=="08", RUN+="usb_modeswitch '%b/%k'"
/var/log/syslog (after sudo udevadm control --log-priority=debug):
Lite: Oct 19 17:10:40 raspberrypi systemd-udevd[705]: starting 'usb_modeswitch '1-1.4/1-1.4:1.0''
Desktop: Oct 19 19:22:26 raspberrypi systemd-udevd[910]: starting 'usb_modeswitch '/1-1.4:1.0''

If I start same command by hand with same parameter as in desktop, it works:
sudo /lib/udev/usb_modeswitch /1-1.4:1.0

I can solve problem by changing '%b/%k' to '%k' in /lib/udev/rules.d/40-usb_modeswitch.rule, but I'm not sure if that it is right solution.

leolobato, 2016-11-09:
I Have a Huawei E7173s and can confirm that on a fresh install of Raspbian Jessie lite (after sudo apt-get update and upgrade), changing `%b/%k` to `%k` on /lib/udev/rules.d/40-usb_modeswitch.rules (line 24) did switch the modem on after a reboot.

popcornmix, 2016-11-10:
This doesn't sound like a linux kernel issue, but a distribution issue.
Reporting this https://github.com/RPi-Distro/repo/issues would be a better place.

closed 2016-11-10 by popcornmix