raspberrypi/linux #1691: Use 2 screens

frank4466, 2016-10-20:
Hi,
I have the raspberry [TouchScreen ](https://www.raspberrypi.org/blog/the-eagerly-awaited-raspberry-pi-display/)and raspberry PI2b+ with pixel OS. I want use 2 screens, the first is HDMI (in VGA mode) and second is Raspberry touchscreen. 
In the article above i can do this with "omxplayer --display=4" command. Is it possible to run a processing sketch on the second screen ? 
Frank

LoyalNine1487, 2016-10-21:
1. This repo is for the kernel and kernel related items only.
2. Go on the Raspberry pi forms [here](https://www.raspberrypi.org/forums/) to ask other questions or report issues that don't have to do with the kernel.

frank4466, 2016-10-21:
Sorry,
I thought it was a kernel modification problem getting a true multi screen functionality.
I post this question on the raspberry forum.

closed 2017-05-04 by P33M