raspberrypi/linux #1694: CVE-2016-5195: Dirty COW

silberzwiebel, 2016-10-21:
Hi,

I'm running raspbian jessie with
```
Linux pi 3.18.0-trunk-rpi #1 PREEMPT Debian 3.18.5-1~exp1+rpi19 (2015-08-08) armv6l GNU/Linux
```
I'm unsure whether this protects me from the recently discovered security relevant bug CVE-2016-5195 (aka dirty COW, [debian bug](https://security-tracker.debian.org/tracker/CVE-2016-5195), [more information](http://dirtycow.ninja/)), since the debian bug does not list the kernel version I am running. Could you give information what kernel version is not affected?

Many thanks,
silberzwiebel

XECDesign, 2016-10-21:
The kernel you are running is not provided by raspberry pi (it comes from the raspbian.org repo). The problem itself has been fixed in this repo (https://github.com/raspberrypi/linux/commit/1294d355881cc5c3421d24fee512f16974addb6c). No idea if or when it will be fixed, since it's outside of our control

The raspberrypi-kernel package, which is distributed by us, has not been updated yet.

ufa, 2016-10-21:
"The raspberrypi-kernel package, which is distributed by us, has not been updated yet."
Is it a WIP?

popcornmix, 2016-10-21:
``rpi-update`` will get you kernel 4.4.26 (which includes the fix), although be aware that is a bleeding edge kernel/firmware.
It will make it into raspberrypi apt-get repo at some point (we want to ensure the bleeding edge firmware/kernel hasn't introduced any regressions).

ufa, 2016-10-21:
Thank you popcornmix

XECDesign, 2016-10-24:
The fix should be in apt now.
```
sudo apt-get update
sudo apt-get dist-upgrade
```

vk5tu, 2016-10-25:
`apt-get update && apt-get dist-upgrade` on Raspbian Wheezy on a RPi1v2 says `Linux pi 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux`.

Can we expect an updated Raspbian kernel for Wheezy?  Or is the rpi-update kernel the best way forward?

XECDesign, 2016-10-25:
In general, we don't support Wheezy. However, this is a bit of a special case. If enough people confirm that everything works fine on Wheezy with the new packages, they may be added to Jessie.

closed 2017-05-18 by P33M