raspberrypi/linux #1695: (Pi2) 4.4.19-v7+ ethernet times out and triggers NETDEV_WATCHDOG

DavidVentura, 2016-10-21:
I've been having random lockups the last few days. This time the pi came back to life and I got the kernel logs.

I have a wireless keyboard dongle and a sound card connected to the pi's usbs. The power supply is 2 amps and there's no low-power "rainbow". I also have the pi-camera connected, this happened **exactly** when I toggled it (on->off->on)

I'm using nfsroot but the filesystem is almost not used. I'm streaming the picam over udp multicast (few hours every day).

```
$ /opt/vc/bin/vcgencmd version
Aug 23 2016 15:58:54 
Copyright (c) 2012 Broadcom
version f23e65aeeffca65654550c93254a0ee55f34fb07 (clean) (release)
```
```
$ uname -a
Linux jessie-rpi 4.4.19-v7+ #906 SMP Tue Aug 23 15:53:06 BST 2016 armv7l GNU/Linux
```

```
[ 5615.036966] ------------[ cut here ]------------
[ 5615.037028] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x2a4/0x2b0()
[ 5615.037043] NETDEV WATCHDOG: eth0 (smsc95xx): transmit queue 0 timed out
[ 5615.037056] Modules linked in: uinput snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device snd_pcm snd_timer snd evdev bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio bcm2835_v4l2 videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media cuse fuse ipv6
[ 5615.037234] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.4.19-v7+ #906
[ 5615.037247] Hardware name: BCM2709
[ 5615.037292] [<80018784>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
[ 5615.037321] [<80014058>] (show_stack) from [<80320c84>] (dump_stack+0xd4/0x118)
[ 5615.037351] [<80320c84>] (dump_stack) from [<80025360>] (warn_slowpath_common+0x98/0xc8)
[ 5615.037378] [<80025360>] (warn_slowpath_common) from [<800253d0>] (warn_slowpath_fmt+0x40/0x48)
[ 5615.037404] [<800253d0>] (warn_slowpath_fmt) from [<804f90d0>] (dev_watchdog+0x2a4/0x2b0)
[ 5615.037431] [<804f90d0>] (dev_watchdog) from [<80081cb0>] (call_timer_fn+0x40/0x1a0)
[ 5615.037454] [<80081cb0>] (call_timer_fn) from [<80081fd0>] (run_timer_softirq+0x1c0/0x324)
[ 5615.037479] [<80081fd0>] (run_timer_softirq) from [<800290b0>] (__do_softirq+0x18c/0x3cc)
[ 5615.037502] [<800290b0>] (__do_softirq) from [<80029690>] (irq_exit+0xdc/0x140)
[ 5615.037527] [<80029690>] (irq_exit) from [<80071174>] (__handle_domain_irq+0x70/0xc4)
[ 5615.037554] [<80071174>] (__handle_domain_irq) from [<80010a74>] (handle_IRQ+0x2c/0x30)
[ 5615.037578] [<80010a74>] (handle_IRQ) from [<8000951c>] (bcm2836_arm_irqchip_handle_irq+0xb8/0xbc)
[ 5615.037606] [<8000951c>] (bcm2836_arm_irqchip_handle_irq) from [<805ba204>] (__irq_svc+0x44/0x5c)
[ 5615.037620] Exception stack(0x80861f08 to 0x80861f50)
[ 5615.037639] 1f00:                   00000000 b20823c8 00000000 00000000 80860000 808625dc
[ 5615.037659] 1f20: 80862500 80862580 805bee5c 808c68d8 b27ff540 80861f64 808634f8 80861f58
[ 5615.037675] 1f40: 80010b34 80010b38 600f0013 ffffffff
[ 5615.037702] [<805ba204>] (__irq_svc) from [<80010b38>] (arch_cpu_idle+0x34/0x4c)
[ 5615.037729] [<80010b38>] (arch_cpu_idle) from [<80063ea0>] (default_idle_call+0x34/0x48)
[ 5615.037756] [<80063ea0>] (default_idle_call) from [<800640c8>] (cpu_startup_entry+0x214/0x2b0)
[ 5615.037782] [<800640c8>] (cpu_startup_entry) from [<805b4ab4>] (rest_init+0x88/0x8c)
[ 5615.037809] [<805b4ab4>] (rest_init) from [<807e9d34>] (start_kernel+0x3dc/0x3e8)
[ 5615.037855] ---[ end trace 90a6e2e679dc87a1 ]---
[ 5616.026435] smsc95xx 1-1.1:1.0 eth0: link down
[ 5617.690953] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[ 5621.557097] nfs: server 192.168.1.123 not responding, still trying
[ 5656.827139] nfs: server 192.168.1.123 not responding, still trying
[ 5656.828221] nfs: server 192.168.1.123 OK
[ 5656.829559] nfs: server 192.168.1.123 OK
```

Ruffio, 2017-02-05:
@DavidVentura is this still an issue?

P33M, 2017-05-04:
Assuming dupe of #1842. Fix committed.

closed 2017-05-04 by P33M