raspberrypi/linux #1697: kgdboc=ttyAMA0 not working on RPi3

HiassofT, 2016-10-22:
While trying to do some kernel debugging I noticed that kgdb/kgdboc wasn't available when booting the kernel on a RPi3 - on a RPi2 it works as expected (sysrq+g available).

I'm using `dtoverlay=pi3-disable-bt` in config.txt and this cmdline.txt to get a serial console via the PL011/ttyAMA0 uart:
```dwc_otg.lpm_enable=0 console=serial0,115200 kgdboc=serial0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait```

I've booted the kernel with initcall_debug and it looks like on the RPi3 ttyAMA0 hadn't been registered by the time init_kgdboc is run - therefore kgdboc isn't registering (returning ENODEV). Here are dmesgs of the current rpi-update kernel:
RPi2: http://paste.debian.net/886822/
RPi3: http://paste.debian.net/886821/

@pelwell @popcornmix any ideas how to work around or fix this? How are you debugging the kernel on RPi3?

Electron752, 2016-11-02:
Hi, I just stumbled on this:

I've also had lots of issues with the serial port on the RPI3.   Sometimes adding this to your config.txt helps:
enable_uart=1
core_freq=250

You may also need to add:
force_turbo=1
arm_freq=1200

pelwell, 2016-11-02:
You should normally(*) never need force_turbo=1 to get a stable uart, and if you do you are safer using `core_freq=250` - Pi3's get hot with force_turbo.

ttyAMA0 is unaffected by CPU clock speed changes, and if ttyS0 is the primary UART (the one aliased to serial0) then `enable_uart=1` implies `core_freq=250`.

I don't use a kernel debugger - I've always managed with printk - but is the parameter perhaps missing a baudrate - `kgdboc=serial0,115200`?

(*) The exception is if you are trying to use ttyS0 as the secondary uart, which would be the case on a Pi1, Pi2, or on a Pi3 with one of the UART-manipulating overlays: pi3-disable-bt, pi3-miniuart-bt.

HiassofT, 2016-11-05:
kgdboc is quite handy to debug OOPSes (kernel thread is automatically halted and you can attach gdb to check what went wrong), for most other stuff I use printk as well.

I did some more testing and found out why it fails on RPi3 but not RPi2 with kernel 4.4:

On RPi3 the uart DT contains a pinctrl reference, but pinctrl-bcm2835 hasn't been inited by the time so the pl011 driver is scheduled for deferred probing.

The reason for this is that the pl011 driver is registered at arch_initcall level but pinctrl-bcm2835  way later at device_initcall. Changing pinctrl-bcm2835 to register at arch level fixes this issue (most other pinctrl drivers use the arch level as well):
```
--- a/drivers/pinctrl/bcm/pinctrl-bcm2835.c
+++ b/drivers/pinctrl/bcm/pinctrl-bcm2835.c
@@ -1117,7 +1117,12 @@ static struct platform_driver bcm2835_pinctrl_driver = {
                .of_match_table = bcm2835_pinctrl_match,
        },
 };
-module_platform_driver(bcm2835_pinctrl_driver);
+
+static int __init bcm2835_pinctrl_init(void)
+{
+       return platform_driver_register(&bcm2835_pinctrl_driver);
+}
+arch_initcall(bcm2835_pinctrl_init);
```

With kernel 4.8 there's another problem: the pl011 isn't registered beecause the amba bus depends on the clock driver, which again uses device initcall level - so devices on the amba bus are sent into deferred probing as well. Again the fix seems to be to register clk-bcm2835 at arch level:

```--- a/drivers/clk/bcm/clk-bcm2835.c
+++ b/drivers/clk/bcm/clk-bcm2835.c
@@ -1909,7 +1909,12 @@ static struct platform_driver bcm2835_clk_driver = {
        .probe          = bcm2835_clk_probe,
 };

-builtin_platform_driver(bcm2835_clk_driver);
+static int __init bcm2835_clk_init(void)
+{
+       return platform_driver_register(&bcm2835_clk_driver);
+}
+
+arch_initcall(bcm2835_clk_init);
```

With these fixes none of the core drivers (stock config.txt with only console=serial0, ... added to cmdline.txt) are running into deferred probing and the serial console is avaiable quite early during boot.

BTW not sure if arch is the correct level for clk (most other clk drivers register at core), maybe other drivers are registering too late as well. 

@pelwell @anholt any ideas?

clivem, 2016-11-05:
I'm not sure why @popcornmix reverted the @msperl patch to get clockmgr registered early.....

popcornmix, 2016-11-05:
The clockmgr register early patch causes an unpleasant glitch on the HDMI display I was using when kernel starts to boot. Not tracked down the issue, but I'm guessing it is affecting PLLs/clocks at a point it shouldn't.

msperl, 2016-11-05:
I believe there was somewhere a clock-patch with regards to HDMI that should solve that issue...

Could be one of the following:
* "clk: bcm2835: Skip PLLC clocks when deciding on a new clock parent"
* "clk: bcm2835: Clamp the PLL's requested rate to the hardware limits."

I guess it is the second one.

JamesH65, 2017-05-18:
@popcornmix Not sure what going on with this one. Next steps?

kaba-official, 2018-05-27:
I'm still facing this issue with branch **rpi-4.9.y**. I could see the early clock issue has be fixed by @msperl . Still if **kgdboc** is set in **cmdline.txt** file, Pi3 is not booting. I tried porting the **pinctrl-bcm2835 patch** proposed by @HiassofT . That ain't help too. Any work-around? Anybody works on this issue?

pelwell, 2018-05-27:
Have you added `dtoverlay=pi3-disable-bt` or `dtoverlay=pi3-miniuart-bt` to config.txt? If not, ttyAMA0 is mapped to the BT modem pins on a Pi3B (and 3B+ and Zero W).

kaba-official, 2018-05-28:
Thanks for replying @pelwell . Below is my cmdline.txt and config.txt. Once I power-on the Pi, I see nothing in serial console. And I have connected HDMI. In the HDMI monitor I'm seeing only a lightning [power] symbol blinking on top-right corner.

```sh
$ sudo cat /media/kaba/raspberrypi/config.txt 
dtoverlay=vc4-fkms-v3d,cma-256,pi3-disable-bt,pi3-miniuart-bt
disable_overscan=1
dtparam=audio=on
device_tree=bcm2710-rpi-3-b.dtb
enable_uart=1
$ 
```

```sh
$ sudo cat /media/kaba/raspberrypi/cmdline.txt 
dwc_otg.lpm_enable=0 dwc_otg.speed=1 debug console=serial0,115200 kgdboc=ttyS0,115200 earlyprintk root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait 
$ 
```

If I remove ``kgdboc`` option from cmdline.txt, everything works fine.

In my case serial0 is mapped to ttyS0. For verification purpose, I have even tried with ``kgdboc=ttyAMA0,115200``. Still same result.

pelwell, 2018-05-28:
```
dtoverlay=vc4-fkms-v3d,cma-256,pi3-disable-bt,pi3-miniuart-bt
```
This syntax doesn't do what you think it does - symbols after the first comma are interpreted as parameters, either for the named overlay or global.
Instead you should use:
```
dtoverlay=vc4-fkms-v3d,cma-256
dtoverlay=pi3-disable-bt
# dtoverlay=pi3-miniuart-bt  - not necessary
```
With these overlays in place, run `raspi-gpio get` to confirm that TXD0 and RXD0 are mapped to GPIOs 14 and 15 only. Then try re-enabling `kgdboc`.

kaba-official, 2018-05-29:
Thanks a lot @pelwell . You have saved my day. It worked with following values.

```sh
root@raspberrypi3-64:~# cat /boot/cmdline.txt 
dwc_otg.lpm_enable=0 dwc_otg.speed=1 debug console=serial0,115200 kgdboc=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait 
```
```sh
root@raspberrypi3-64:~# cat /boot/config.txt 
disable_overscan=1
dtparam=audio=on
device_tree=bcm2710-rpi-3-b.dtb
dtoverlay=vc4-fkms-v3d,cma-256
dtoverlay=pi3-disable-bt
enable_uart=1
root@raspberrypi3-64:~#
```

JamesH65, 2018-06-28:
@pelwell @kaba-official Looks like this issue can be closed?

closed 2018-06-28 by pelwell