raspberrypi/linux #1701: read_current_timer() is gone on recent kernels. What has replaced it?

orayb, 2016-10-28:
I had been using the read_current_timer() in version 3.12.26 kernel to mark time with 1us resolution. Now that I installed  version 4.4.27+ kernel, I see that the read_current_timer() has been effectively disabled since arch/arm/mach-bcm2708/bcm2708.c is no longer registering a handler for it. Is there an alternate way of marking time at this resolution with the newer kernels?

pelwell, 2016-10-28:
Use `getnstimeofday`. If you look at the [lirc_rpi driver](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/staging/media/lirc/lirc_rpi.c) you'll find a function - [`read_current_us`] (https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/staging/media/lirc/lirc_rpi.c#L110) - that uses `getnstimeofday` to calculate and return time in microseconds.

kukabu, 2016-10-28:
read_current_us is a ugly example. getnstimeofday uses CLOCK_REALTIME scale in such a manner time can jumps forward/backward.
use ktime_get_ts(&ts) for kernel and clock_gettime(CLOCK_MONOTONIC, &ts) for userspace or other monotonic scales.

orayb, 2016-10-28:
Pelwell and Kukabu, thanks for the feedback. I now have something to work with.

closed 2016-10-28 by orayb