raspberrypi/linux #1702: how to run teamspeak 3 server in raspberry pi

Kimbyeoungjang, 2016-10-28:
i have a raspberry pi 3. i want to run teamspeak 3 server

so i use qemu

apt-get install git zlib1g-dev libsdl1.2-dev

and download qemu 0.15.1.tar.gz

./configure --target-list="i386-softmmu" --enable-sdl --extra-cflags="-O3 -mfloat-abi=hard -mfpu=vfp -mcpu=arm1176jzf-s -mtune=arm1176jzf-s -march=armv6zk" --audio-drv-list="alsa oss sdl pa" --audio-card-list="ac97 es1370 sb16 cs4231a adlib gus hda"

make

but i find some error

  LINK  i386-softmmu/qemu
../slirp/misc.o: In function `memset':
/usr/include/arm-linux-gnueabihf/bits/string3.h:81: warning: memset used with constant zero length parameter; this could be due to transposed parameters
/usr/bin/ld: ../qemu-timer.o: undefined reference to symbol 'timer_create@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/librt.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Makefile:398: recipe for target 'qemu' failed
make[1]: *** [qemu] Error 1
Makefile:84: recipe for target 'subdir-i386-softmmu' failed
make: *** [subdir-i386-softmmu] Error 2

please help my

P.s sorry my bad english

pelwell, 2016-10-28:
Your problem isn't a kernel issue so this is the wrong forum - the Raspberry Pi Advanced users forum (https://www.raspberrypi.org/forums/viewforum.php?f=29) would be better.

Fortunately I have a degree in Googling the error message - try this: http://stackoverflow.com/questions/18286738/undefined-reference-to-symbol-timer-settimeglibc-2-3-3

Alternatively, I found this tutorial: https://eltechs.com/run-teamspeak-3-server-on-raspberry-pi/
Note: This page is effectively an advert for ExaGear desktop, which is commercial software, but it's not expensive and may be worth considering.

Disclaimer: I have no relationship to Eltechs, and I hadn't even heard of ExaGear desktop before this issue, so this is not a recommendation.

Kimbyeoungjang, 2016-10-29:
okay thank you very so much pelwell
i solve first question but i have another error

i do "make"

  LINK  i386-softmmu/qemu
../slirp/misc.o: In function `memset':
/usr/include/arm-linux-gnueabihf/bits/string3.h:81: warning: memset used with constant zero length parameter; this could be due to transposed parameters

how to solve this error?

if i solve this error, i can run programs please help me

Kimbyeoungjang, 2016-10-29:
oh i solve problum, thank you every guys

closed 2016-10-29 by Kimbyeoungjang