raspberrypi/linux #1706: build error

goog, 2016-11-02:
make:
```bash
drivers/staging/media/lirc/lirc_rpi.c:392:10: error: dereferencing pointer to incomplete type
  gpiochip->set(gpiochip, gpio_out_pin, invert);
          ^
drivers/staging/media/lirc/lirc_rpi.c:394:20: error: dereferencing pointer to incomplete type
  irq_num = gpiochip->to_irq(gpiochip, gpio_in_pin);
                    ^
drivers/staging/media/lirc/lirc_rpi.c:409:16: error: dereferencing pointer to incomplete type
    if (gpiochip->get(gpiochip, gpio_in_pin))
                ^
drivers/staging/media/lirc/lirc_rpi.c: In function ‘lirc_write’:
drivers/staging/media/lirc/lirc_rpi.c:498:10: error: dereferencing pointer to incomplete type
  gpiochip->set(gpiochip, gpio_out_pin, invert);
          ^
cc1: some warnings being treated as errors
make[4]: *** [drivers/staging/media/lirc/lirc_rpi.o] Error 1
make[3]: *** [drivers/staging/media/lirc] Error 2
make[2]: *** [drivers/staging/media] Error 2
make[1]: *** [drivers/staging] Error 2
make: *** [drivers] Error 2
```

pelwell, 2016-11-02:
You don't say which kernel version you are building, preferably with a commit id, and what commands you ran to build the kernel. However, given that I have just built rpi-4.4.y, rpi-4.7.y, rpi-4.8yy and rpi-4.9.y, I think you are doing something wrong.

We recommend you follow the kernel building instructions [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md).

Electron752, 2016-11-02:
@goog: Can you please add which architecture you are building for and if you are using the default configs?  arm6l+VFP, armhf, arm64, etc?

I recently was building arm64 of the rpi-4.9.y tree and I had to add GPIOLIB explicitly to my config because I was getting very similar errors to this. So if you are not using the default configs you may have problems.

CONFIG_GPIOLIB=y

Please note that arm64 is not really supported at this time.  If you are building that you are on the bleeding edge...

Electron752, 2016-11-02:
@pelwell: I think the problem is that linux/gpio.h and a few other include files recently got some #ifdef defined(CONFIG_GPIOLIB) that were not around before.   I'm wondering if some of the Kconfig files are missing a selects or depends on GPIOLIB?

Electron752, 2016-11-02:
drivers/staging/media/lirc/Kconfig:
config LIRC_RPI
	tristate "Homebrew GPIO Port Receiver/Transmitter for the RaspberryPi"
	**depends on LIRC**
	help
	  Driver for Homebrew GPIO Port Receiver/Transmitter for the RaspberryPi

popcornmix, 2016-11-02:
On 4.4 we have ``CONFIG_ARCH_REQUIRE_GPIOLIB=y`` which doesn't allow it to be disabled.
This will be different on newer kernels.

goog, 2016-11-04:
@pelwell  linux-rpi-4.4.y, the lastest code zip

goog, 2016-11-04:
@pelwell after following your guide link, thing is ok. thx!

closed 2016-11-04 by goog

Gerharddc, 2016-11-28:
@popcornmix for some reason CONFIG_ARCH_REQUIRE_GPIOLIB isn't working on 4.9 for ARCH_BCM2709 in Kconfig.platforms for aarch64 because GPIOLIB comes up as disabled in the config file and adding GPIOLIB manually works. Any reason why?