raspberrypi/linux #1708: 4.4.26 WARNING: CPU: 1 PID: 0 at kernel/rcu/tree.c:2733 rcu_process_callbacks+0x6f8/0x738()

brent80, 2016-11-03:
```kernel: [44238.769123] ------------[ cut here ]------------
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769178] WARNING: CPU: 1 PID: 0 at kernel/rcu/tree.c:2733 rcu_process_callbacks+0x6f8/0x738()
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769191] Modules linked in: tun sha256_generic hmac drbg md5 ecb cfg80211 rfkill ip6table_filter ip6_tables iptable_filter ipt_MASQUERADE nf_nat_masquerade_ipv4 xt_nat xt_tcpudp iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_tables snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio i2c_dev fuse ecryptfs ipv6
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769380] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.4.26-v7+ #915
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769393] Hardware name: BCM2709
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769435] [<80018784>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769465] [<80014058>] (show_stack) from [<80321364>] (dump_stack+0xd4/0x118)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769496] [<80321364>] (dump_stack) from [<800253c4>] (warn_slowpath_common+0x98/0xc8)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769525] [<800253c4>] (warn_slowpath_common) from [<800254b0>] (warn_slowpath_null+0x2c/0x34)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769552] [<800254b0>] (warn_slowpath_null) from [<8007b950>] (rcu_process_callbacks+0x6f8/0x738)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769580] [<8007b950>] (rcu_process_callbacks) from [<80029114>] (__do_softirq+0x18c/0x3cc)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769605] [<80029114>] (__do_softirq) from [<800296f4>] (irq_exit+0xdc/0x140)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769631] [<800296f4>] (irq_exit) from [<80071320>] (__handle_domain_irq+0x70/0xc4)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769658] [<80071320>] (__handle_domain_irq) from [<80010a74>] (handle_IRQ+0x2c/0x30)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769683] [<80010a74>] (handle_IRQ) from [<8000951c>] (bcm2836_arm_irqchip_handle_irq+0xb8/0xbc)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769712] [<8000951c>] (bcm2836_arm_irqchip_handle_irq) from [<805baac4>] (__irq_svc+0x44/0x5c)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769727] Exception stack(0xb993bf48 to 0xb993bf90)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769746] bf40:                   00000000 b9f6f3c8 00000000 00000000 b993a000 808625dc
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769767] bf60: 10c0387d 80862580 805bee5c 808c6918 00000000 b993bfa4 808634f8 b993bf98
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769782] bf80: 80010b34 80010b38 60000013 ffffffff
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769810] [<805baac4>] (__irq_svc) from [<80010b38>] (arch_cpu_idle+0x34/0x4c)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769838] [<80010b38>] (arch_cpu_idle) from [<8006404c>] (default_idle_call+0x34/0x48)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769866] [<8006404c>] (default_idle_call) from [<80064274>] (cpu_startup_entry+0x214/0x2b0)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769892] [<80064274>] (cpu_startup_entry) from [<80015c88>] (secondary_start_kernel+0x12c/0x138)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769915] [<80015c88>] (secondary_start_kernel) from [<000095ac>] (0x95ac)
Nov  3 01:01:12 lumberJack-02 kernel: [44238.769930] ---[ end trace 5b03a45f64cb8713 ]---```

brent80, 2016-11-03:
This is a raspberry pi 2 image running on a pi3, could this be part of the issue?

pelwell, 2016-11-03:
What do you mean by "pi 2 image"? A standard RPi firmware uses the same kernel on Pi2 and Pi3 - where did this kernel come from?

brent80, 2016-11-03:
Well it just started out on a pi2, Made an image of the hdd and put it on the new pi 3...I knew they used the same kernel, but I didn't know if there were any difference that could cause this is issue.

pelwell, 2016-11-03:
OK - the build number matches up with rpi-update firmware commit 849dc3a5c3e381991eba31adecd00553a32cb0f8, so I'll assume that's what you are running.

What was the Pi doing when you got the warning?

brent80, 2016-11-03:
Well, it is a remote device.  It just does the same thing day in and day out.  Talk to devices on the network and send out information.  It has a watchdog setup.  But when I got this warning one stopped functioning correctly.  It said illegal character and stopped the process until I rebooted the device.

pelwell, 2016-11-03:
What is the power supply rated at? What does `vcgencmd get_throttled` return? Random failures like this in what should be solid kernel code usually point to under-voltage problems.

brent80, 2016-11-03:
throttled=0x0, 5v 3amp, but the power grid is not the best.  It is in a remote area and it could be possible that the voltage drops.  I have no way to verify that.

pelwell, 2016-11-03:
If we assume that there would be some non-fatal power dips as well as fatal ones then periodically calling vcgencmd get_throttled and logging the output to disk might catch a problem. Or not, if there isn't one.

brent80, 2016-11-03:
I will try to update you in the future, if i get some results with that command.

brent80, 2016-11-04:
@pelwell 
Had this happen again on a system I have locally, it has good power.  This has been running this kernel for awhile, the only things that have changed are adding a watchdog and updating a program, maybe our program is doing something bad, but I disabled the watchdog.  Let you know
```
raspberrypi-02 kernel: [509490.688179] ------------[ cut here ]------------
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688222] WARNING: CPU: 0 PID: 33 at fs/seq_file.c:65 seq_open+0x9c/0xa8()
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688230] Modules linked in: tun sha256_generic hmac drbg md5 ecb cfg80211 rfkill ip6table_filter ip6_tables iptable_filter ipt_MASQUERADE nf_nat_masquerade_ipv4 xt_nat xt_tcpudp iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_ta$
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688348] CPU: 0 PID: 33 Comm: kswapd0 Not tainted 4.4.26-v7+ #915
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688355] Hardware name: BCM2709
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688384] [<80018784>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688403] [<80014058>] (show_stack) from [<80321364>] (dump_stack+0xd4/0x118)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688421] [<80321364>] (dump_stack) from [<800253c4>] (warn_slowpath_common+0x98/0xc8)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688439] [<800253c4>] (warn_slowpath_common) from [<800254b0>] (warn_slowpath_null+0x2c/0x34)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688455] [<800254b0>] (warn_slowpath_null) from [<8017b430>] (seq_open+0x9c/0xa8)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688471] [<8017b430>] (seq_open) from [<8017bd34>] (single_open+0x78/0xa8)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688489] [<8017bd34>] (single_open) from [<8007e800>] (prof_cpu_mask_proc_open+0x28/0x2c)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688507] [<8007e800>] (prof_cpu_mask_proc_open) from [<80113d00>] (shrink_page_list+0x750/0xa70)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688524] [<80113d00>] (shrink_page_list) from [<80114614>] (shrink_inactive_list+0x1d4/0x520)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688539] [<80114614>] (shrink_inactive_list) from [<80115214>] (shrink_lruvec+0x560/0x698)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688554] [<80115214>] (shrink_lruvec) from [<80115454>] (shrink_zone+0x108/0x2f0)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688569] [<80115454>] (shrink_zone) from [<80116918>] (kswapd+0x668/0xa3c)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688584] [<80116918>] (kswapd) from [<80042860>] (kthread+0xec/0x104)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688600] [<80042860>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
Nov  3 15:39:13 raspberrypi-02 kernel: [509490.688610] ---[ end trace c95da4b7cae80b78 ]---
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715443] INFO: rcu_sched self-detected stall on CPU
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715488]  0-...: (2099 ticks this GP) idle=c0b/140000000000001/0 softirq=6316173/6316173 fqs=2099
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715498]   (t=2100 jiffies g=4433017 c=4433016 q=372)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715511] Task dump for CPU 0:
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715518] kswapd0         R running      0    33      2 0x00000002
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715559] [<80018784>] (unwind_backtrace) from [<80014058>] (show_stack+0x20/0x24)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715578] [<80014058>] (show_stack) from [<8004febc>] (sched_show_task+0xb4/0x108)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715595] [<8004febc>] (sched_show_task) from [<80052190>] (dump_cpu_task+0x48/0x4c)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715612] [<80052190>] (dump_cpu_task) from [<80079d7c>] (rcu_dump_cpu_stacks+0x9c/0xd8)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715629] [<80079d7c>] (rcu_dump_cpu_stacks) from [<8007ddd0>] (rcu_check_callbacks+0x5c0/0x904)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715646] [<8007ddd0>] (rcu_check_callbacks) from [<80083128>] (update_process_times+0x48/0x74)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715662] [<80083128>] (update_process_times) from [<800945f4>] (tick_sched_handle+0x58/0x64)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715677] [<800945f4>] (tick_sched_handle) from [<80094658>] (tick_sched_timer+0x58/0x9c)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715693] [<80094658>] (tick_sched_timer) from [<80083ce8>] (__hrtimer_run_queues+0x170/0x348)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715707] [<80083ce8>] (__hrtimer_run_queues) from [<80084730>] (hrtimer_interrupt+0xd4/0x230)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715725] [<80084730>] (hrtimer_interrupt) from [<8048ea74>] (arch_timer_handler_phys+0x40/0x48)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715744] [<8048ea74>] (arch_timer_handler_phys) from [<800759b4>] (handle_percpu_devid_irq+0x80/0x194)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715761] [<800759b4>] (handle_percpu_devid_irq) from [<80071010>] (generic_handle_irq+0x34/0x44)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715777] [<80071010>] (generic_handle_irq) from [<8007131c>] (__handle_domain_irq+0x6c/0xc4)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715793] [<8007131c>] (__handle_domain_irq) from [<80010a74>] (handle_IRQ+0x2c/0x30)
[... cut ...]
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715966] [<80114614>] (shrink_inactive_list) from [<80115214>] (shrink_lruvec+0x560/0x698)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715982] [<80115214>] (shrink_lruvec) from [<80115454>] (shrink_zone+0x108/0x2f0)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.715996] [<80115454>] (shrink_zone) from [<80116918>] (kswapd+0x668/0xa3c)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.716012] [<80116918>] (kswapd) from [<80042860>] (kthread+0xec/0x104)
Nov  3 15:39:34 raspberrypi-02 kernel: [509511.716027] [<80042860>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
```

Ruffio, 2017-02-05:
@brent80 can this be closed?

brent80, 2017-02-05:
I guess, I switched to ubuntu server 16.04 and this problem went away.  If no one else is having this issue,  maybe my SD card had some corruption or something.

closed 2017-02-05 by brent80