raspberrypi/linux #1709: USB (VCP) communication issues

carmarri, 2016-11-03:
Hello,

I've noticed some issues in the USB communication. I'll try to give us much information as possible
More information (data, pictures, etc) on previous discussion on Raspberry PI forum: [](https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=161922&p=1062581#p1062581)

**Introduction**
I'm developing a SO library (C/C++) for one of my company products, using codeblocks and cross-compiling with GCC from Ubuntu Mate. This hardware product uses FTDI and is connected to the Raspberry PI using a USB cable. At the software layer, the communication is performed using the Virtual Comm Port, since I want to allow the Library to detect any port, no matter if its VCP or a physical port. In order to handle the communications I use the kernel library commands open(), close(), read(), write(), ioctl(), termios structures, etc.
The Library has been tested from the raspberry PI through Lazarus (free-pascal) and Python, having both the same behaviour/issue.

**Issue**
If I open the port, send and receive data to/from my device, the communication runs perfect until I close the port. So what I've noticed is that, randomly (maybe once every 5 times), when I try to open the port, the port is succesfully opened (all functions return succeed) but no communication can be performed. I mean, Once the port is open, I send a fixed ASCII command to our product and I expect a fixed response (product detection through an ID). The command is sent sucessfully, in fact, no errors are reported by the write() function, then I've a timeout in the response and I don't get any data. The command should be received within 20-30ms and I've a timeout of 1 second, which I've experimentally increased during testing up to 10 seconds. The bytes available in the buffer/ports are zero. Then, no matter if I send more commands or not, the communication still fails.
So if I open the port and I'm able to communicate, everything goes fine until I close the port; but if the first command fails, then the communication keeps failing until I close the port.
I've monitored the TTL serial lines after the FTDI in my device (see link) and I've noticed that when the communication fails at the software layer, it is actually being performed at the hardware layer.
I'm using a Raspbian Lite Image to which I've installed LXDE and dependencies, since the application is targeting an embedded touch-screen system.

**Testing/Debug**
- The Raspberry PI is up to date, all packages are up to date.
- Tested on a dirty and clean Debian/Raspbian update.
- My device has both USB and RS232 ports, so I've tryed connecting to the physical serial port of the Raspberry PI and the SO library works fine 100% of times, so it always connects and communicates fine.
- I've tested other USB to RS232 chips and converters and the same error is repeated.
- The same hardware has been tested from a PC and no problems are reported, so is not, an isue of the hardware.
- Different baudrates have been tested and the problem still occurs.
- The very same SO code has been compiled and tested on a Linux desktop PC and everything works fine, so it's not an issue of the Library.
- The very same library and test programs have been tested in Raspberry PI v3, v2 and v1 and the same problem occurs, so the problem is repeatable and consistant.
- The very same SO and test programs (copy-paste) have been tested in other ARM platforms, also with Debian: BeagleBone black, Cubie-board, UDOO Neo. and everything works fine, the problem does not exists, so clearly it is related to the Raspberry PI. Based on this test I suppose that the problem is not in the FTDI or VCP drivers, since I guess the drivers are the same and used along different distributions and platforms (am I right?).

I'm not sure what could be the problem but maybe there is a problem while handling the USB interrupts (kernel issue); I initially though that could be something related to UART syncronization but this does not make sense in USB Virtual Comm Ports, since everything is software, handled by the USB driver.

It took me one month of testing before I had enough data to report this, so please, take this seriously.

Since I work indirectly for the automotive industry and there are many companies all over the globe pretending to develop projects for the Raspberry PI, it will be a pitty that the Raspberry PI is discarted or not recommended if they start noticing this kind of issues.

Please, feel free to ask for more data if you need it.

Thank you very much in advance.

pelwell, 2016-11-03:
You will have a better chance of getting to a solution if you cut your application down to a small test program that still exhibits the problem and upload the source somewhere - a Gist, Pastbin, Google Drive etc. - together with any instructions you think are relevant. An ideal test would involve looping back the USB serial port to the onboard UART, with a small application to emulate your device.

P33M, 2016-11-03:
It would also be useful if we had an indication of how often this problem occurs - is it 1 time in 10? 10000? Does it recur within seconds or hours?

In addition to cutting the application down to minimal source/test harness, finding out what provokes the bug more reliably (e.g. repeatedly opening/closing the device in a naive fashion) would make the likelihood of finding it a lot easier.

carmarri, 2016-11-03:
Hi,

@P33M  I've already given that info in my post. It is completely random, but it usually happen 1 time in 5-10.

What it is very important and I forgot to add is (I'll edit the post):
- The error does not occur if I repeat the sequence open-communicate-close continuously. I mean, I have a button that I click every time I want to open and close the communication. If I open the port immediately that it has been closed, the issue is minimized and rarely occurs. But if I wait for some seconds after I close the port, then the issue occurs.

@pelwell I'll discuss this with my boss and I'll see what I can do. I guess that I can post something, since the code is very similar to many Serial port communication codes for C/C++ in Linux, that can be found on the net. I guess that you would need a similar setup, but this is delicate. I guess that I would be allowed to send a demo version of the hardware with an NDA to somebody at the Raspbeery PI foundation.

pelwell, 2016-11-03:
If the problem is in the USB serial port side as you claim then a simple test harness that opens a serial port on the Pi (/dev/serial0), waits for the greeting and sends the response, ought to be adequate - just attach the pins on the FTDI to header pins 6, 8 and 10 (you don't say if you are using flow control).

Electron752, 2016-11-05:
@carmarri:

If you suspect this is a USB issue, you might want to see if you can get a trace with usbmon.  It's a free driver that's included with Linux that let's you snoop the USB information going into and out of the USB driver.

Also, if your company have a few $$$ at your disposal, you can try buying a USB bus analyzer.   These can be on the pricey side($1,000) but can be very useful for these kinds of issues.  Some standard logic analyzer's have some USB support, but they are very limited compared to an actual bus analyzer.

Finally, I think some of FTDI's products are garbage even though they seem to be very, very popular.   I tried using them 15 year ago for something and I was having all kinds of issues similar to yours.   I think they have improved somewhat since then though.

lategoodbye, 2016-11-05:
@carmarri Does dmesg show anything in case the issue occurs?

carmarri, 2016-11-14:
Hi,

First of all, sorry for the delay, but have been out of the country for the last week on a work trip.

@pelwell As I said on the initial message, I already tested communication through the physical UART ports using the very same software/settings and it is working fine. The problem is on the USB. My device is USB based (FTDI) so it is connected to the Raspberry PI using a USB cable.

@Electron752 Thanks for the suggestion. I'll try usbmon. I don't think my boss will be happy to buy a USB Analyzer to debug this issue. Anyway I think this is not related to FTDI, since, as I said in my initial message, I've tried different serial to USB converters with the very same result (Cypress, SiLabs and another Chinesse product). FTDI is probably not the best solution (I don't have much experience with others) but it has been reliably working for the last 8 years. Do you have any suggestion to use a different brand?

@lategoodbye dmesg does not show any message  when the error occurs. The messages available are from the boot sequence:
registered new interface usbserial
registered new interface usbserial_generic
USB Serial support registered for generic
registered new interface driver for ftdi_sio
USB Serial support registered for FTDI USB Serial Device
FTDI USB Serial Device converter detected
Detected FT232RL
FTDI USB Serial Device converter now attached to ttyUSB0

*** As a sumary of the issue: My device can be connected to the Raspberry PI using either USB or serial cable. With Serial connected to the physical UART pins of the PI, no error detected. When connected using USB the problem occurs. The problem does occurs on board v1, v2 and v3, but not on other boards like BeagleBone, UDOO, etc. I've monitored the serial lines in my device (before the FTDI chip) and when the issue appears the command is received from the computer and a response is sent to it, but an empty buffer is reported on the C++ code.

Regards,
Carles

pelwell, 2016-11-14:
@carmarri I think you have misunderstood. I am suggesting that you can use a working "physical" UART to emulate the greeting handshake in your device, which you then connect to the FTDI USB UART:

```
Pi application -> FTDI UART -> Pi onboard UART -> simple emulated device
```

carmarri, 2016-11-14:
Hi @pelwell sorry but I indeed missunderstood what you meant. I guess that you want to confirm that this is not a hardware problem. Well, I can confirm you that it is not. since it has been tested with other FTDI-based product and the same problem occurs, while in other boards does not occur. Also tested this hardware in x86 Laptop/Desktop PCs and the problem does not occur either.

pelwell, 2016-11-14:
No, that isn't what I want. I believe that you are experiencing a real problem, I am assuming for now that it isn't caused be a faulty device, and I would like to be able to reproduce the problem here.

Your options are:

1. Send us all of the hardware and application software discussed. This could involve NDAs and lawyers.
2. Send us the hardware and some simple test code. Perhaps not as bad as 1, but still not great.
3. Demonstrate that the same problem can be observed with an FTDI cable and an emulator running on a Pi. All you then need to send us is the test app and an FTDI cable, but we may even be able to observe it with one of the many cables we already have.

I assumed that option 3 would be easier all round, so that is what I have been advocating.

carmarri, 2016-11-16:
Hi @pelwell 
I've just tested what you suggested in 3). I've created a c++ program that emulates the behaviour of our hardware and I've it running on a Raspberry PI v1, then I'm using a FTDI serial to USB cable connected to the physical serial port of the first PI and connected through USB to the second PI. So the second PI (v3) is running my test application (the same I was using in the inital test) and the same behaviour occurred.
Just in case I did any mistake on the C/C++ code of the test application or the emulator I've tryed creating both emulator and test program in Python too, but with the very same beahavior/issues. So definitely, this is nothing to do with my code. or my hardware.

I'll arrange some test code for you. Please, send me a p.m. with an email address. This is a little bit embarrassing, but my boss is also asking for Name, work address, or some ID that could identify you.

@Electron752 I've tested usbmon, but I'm not sure how to understand what is going on. I've noticed different set of messages between both situations (port opened and communicating vs opened and no communication); the main thing I've noticed is that when everything goes fine there is some kind of periodic control messages on the USB bus, no matter if its transmiting useful data or not (this does not happen when the issue occurs). So I guess, that for some reasone, the USB communication is not being started, but no error is reported to the kernel, so open() returns "no error".

pelwell, 2016-11-16:
GitHub doesn't do PM, but if you email me (phil@raspberrypi.org) saying what information you need I'll get back to you.

carmarri, 2016-11-16:
Thanks @pelwell I'll contact you tomorrow. I really appreciate your time and patience :-)

pelwell, 2016-11-24:
With the help of the supplied test code we've been able to reproduce the problem, capture traces using a USB analyzer and the kernel usbmon facility. Both show long periods where the device isn't polled for data, and if we don't poll it can't send anything. These gaps correspond to the stalls in the data flow as seen by the test client, so there is clear evidence of a bug.

The resident expert on the USB stack (@P33M) is looking forward to opening that can of worms again.

P33M, 2017-04-24:
@carmarri 
Do you still see the issue if you set the parameter `dwc_otg.nak_holdoff=0` in /boot/cmdline.txt? Your test program passes for me with this set. Note that this will degrade performance due to excessive interrupts as it's a workaround (or confirmation of where to look for a bug) rather than a fix.

P33M, 2017-04-24:
I now have both usbmon and analyzer traces for each side of the interface (host driver vs actual hardware).

It appears that a bulk URB is submitted to the HCD for the IN endpoint of the FTDI device, but the USB transfer doesn't take place for upwards of 700ms. The completion callback is then triggered. Prior to this, a pile of transfers were submitted to the control endpoint which presumably form part of the usb-serial driver's initialisation of the device.

There's only a few places in the driver where nak_holdoff can alter the state flow so this narrows down the scope of investigation considerably.

P33M, 2017-04-25:
The cause of the issue has been identified - our nak_holdoff mechanism doesn't take into account URBs being removed from the queue as a trigger to reset the qh->nak_frame count.

The hold-off records the last frame number that a NAK response was received, plus a configurable interval. The queue of subsequent transfers to the endpoint would then be ignored by the driver until (frame_number + interval) had expired.

Long story short, this could cause delays of up to 2 seconds in getting data from a bulk or control full- or low-speed IN endpoint as a stale qh->nak_frame value would erroneously throttle the endpoint by skipping the endpoint queue for a large interval until the frame counter wrapped back around to the nak_frame value. Setting nak_holdoff=0 avoided this path as queues are unconditionally executed.

I've got a 2-line fix that appears to resolve the issue in the test program. Will push this in a PR shortly.

carmarri, 2017-04-26:
Hi! I'm really sorry for the delay in my reply but I'm just back to the office after having taken some days off.
First of all, THANK YOU VERY MUCH, this is a big improvement.

Phil @P33M, do I need to change the value  "dwc_otg.nak_holdoff=0" in my system or will that be internally fixed in the next system update? Based on your words I guess this was only a patch to bypass the issue.

P33M, 2017-04-26:
Setting `dwc_otg.nak_holdoff=0` is suboptimal for a lot of use-cases. It's a workaround until the fix is available (first in rpi-update firmware in a few days, then in a few weeks as a Raspbian package).

P33M, 2017-04-27:
I've got a fix for the first instance of the problem (high latency when requesting data on an open() and read() call) and I think I'm now seeing
>So if I open the port and I'm able to communicate, everything goes fine until I close the port; but if the first command fails, then the communication keeps failing until I close the port.
in the test program, i.e. you open the port, send a command, the server sees the command and responds (via the USB uart) but you never get any data out of your read() call.

This seems to be happening once every 40-50 times the port is opened, rather than the deterministic behaviour seen previously.

P33M, 2017-04-28:
I've seen an instance of the correct response being sent across the wire after the first message is sent after an open(), but the first URB returned to usb-serial after an open() is just the FTDI 2-byte status packet. Further investigation required.

P33M, 2017-04-28:
the response faile

P33M, 2017-05-02:
I believe I now have a fix for the failure on open().

The issue was caused by 
a) a dequeue being issued to an endpoint that was in the middle of a FIQ-accelerated transfer
b) the last transfer to an endpoint returned data
c) the first transfer data from endpoint on the next open() consisting of both 2-byte status and valid data

In b) the driver didn't save the endpoint data toggle state correctly. Thus the next transfer in c) would get dropped on the floor, and if it contained actual serial port data then that data got lost.

I also found an issue regarding uninitialised packet counts in the host channel structure, where stale hc->start_pkt_count values would result in bogus lengths being passed back to the ftdi_sio driver on transfers that had data toggle errors.

The test program now runs for >10mins with no errors. I will create a PR shortly.

closed 2017-05-08 by P33M

roerch, 2017-07-31:
Sorry to dig up this issue again, but it seems this fix broke something for me. Related post on forum is here 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=188617 but a short summary following:

RPi1 Model B, with some USB devices connected. After firmware update my webcam stopped working (took one picture, then it was removed from USB with
`usb 1-1.3.2: device descriptor read/64, error -32`
`usb 1-1.3-port2: unable to enumerate USB device`.

After getting a new hub (old one was Single TT, new one Multi TT) still didn't solve the problem, I used rpi-update to find the exact version where it was working before. And now I found that commit a86bfee5b47a74c13056997f1e4d8b9d8090b398 broke it. 
However, I don't know how to further debug this problem and help would be appreciated as I don't want to stay on kernel 4.9.24 forever. If you need some detailed output, please let me know.

P33M, 2017-07-31:
Please open a new issue for this regression and in that issue:
- Post the full contents of dmesg since boot, as this will tell me what driver is in use for the device
- Post the full output of `sudo lsusb -v`

rocrail, 2017-11-24:
Hi,

using Raspbian Stretch on a Raspi v2, up to date, there is still an issue with the USB Serial support.
Its impossible to communicate with a connected FTDI device. Sometimes it works for a minute or less, but after a close and open communication is a mess.
On a Raspi v1, Wheezy, it works like a charm.

rocrail, 2017-11-25:
Hi, I tested today with the same SD-Crad on a Raspberry Pi v3.
Same issue. The USB RS232 driver is not usable. The hardware is OK because other none RS232 devices are working fine on the same USB ports.
USB-RS232 support is important for this platform which is commonly used for automation tasks.