raspberrypi/linux #1711: lirc_rpi gpio_in_pull parameter is ignored by module and only useful via device tree

dreamlayers, 2016-11-05:
For lirc_rpi it is possible to use internal pull-up or pull-down on the LIRC input GPIO pin. The module accepts a gpio_in_pull parameter, with "(0 = off, 1 = up, 2 = down, default down)". However, this parameter is ignored. It seems like the module code does absolutely nothing with this parameter. It has a [comment](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/staging/media/lirc/lirc_rpi.c#L357) saying "Because of the lack of a setpull function, only support pinctrl-bcm2835 if using device tree.".  I only got pull up to work by adding a `dtparam=gpio_in_pull=up` line after the `dtoverlay=lirc-rpi` line in /boot/config.txt. This still leaves /sys/module/lirc_rpi/parameters/gpio_in_pull set to the default of 2, but that's irrelevant; pull up is actually enabled.

Another possible workaround is to turn on the pull-up in Python, using something like:
```Python
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18,GPIO.IN,pull_up_down=GPIO.PUD_UP)
```

I feel this is a bug because accepting and then ignoring a module parameter is misleading. People will waste time on this. Documenting the issue in a module source comment is insufficient.

pelwell, 2016-11-06:
Bizarrely, the standard Linux GPIO API does not include a way of setting pulls dynamically. Instead one has to set pulls using the pinctrl mechanism, declaring the pin requirements for device using Device Tree. The only runtime pinctrl interface is the selection of a named state for  a set of pins - as far as I know this feature isn't used within our tree, but it should work.

The old bcm2708_gpio driver had a non-standard "setpull" method, but now we are using the upstream driver the old downstream implementation has been removed from the tree. The Python RPi GPIO library directly manipulates the GPIO hardware, rather than go through a kernel driver. The raspi-gpio utility works the same way and has the same capabilities.

The only way I can resolve this issue without adding lots of non-standard downstream code is to delete the vestigial module parameter and any remaining references to the bcm2708 driver.

pelwell, 2016-12-05:
I've deleted the module parameter. The bcm2708 and general gpiochip cleanup can be done separately.

closed 2017-01-06 by pelwell