raspberrypi/linux #1713: dwc2 gadget mode.

ED6E0F17, 2016-11-08:
Gadget mode broke (for me) sometime after 4.8-rc   
(not tested using defconfig, but Google suggests the error exists upstream: "http://www.spinics.net/lists/linux-usb/msg141067.html" )

[    1.534204] dwc2 20980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    1.535001] dwc2 20980000.usb: DWC OTG Controller
[    1.535083] dwc2 20980000.usb: new USB bus registered, assigned bus number 1
[    1.535158] dwc2 20980000.usb: irq 33, io mem 0x00000000
[    1.535494] usb usb1: Manufacturer: Linux 4.8.6-zero+ dwc2_hsotg
[    1.542460] dwc2 20980000.usb: bound driver g_cdc
[    1.785809] dwc2 20980000.usb: new device is high-speed
[    1.864311] dwc2 20980000.usb: dwc2_hsotg_enqueue_setup: failed queue (-11)

-11 (eagain) suggests that the controller is suspended in DWC2_L2 mode: https://github.com/raspberrypi/linux/blob/24e62728b3fc4f118c8ae17b374bce189bb188fc/drivers/usb/dwc2/gadget.c#L793  "

I will try this patch before trying to bisect two months worth of kernel patches:
http://www.spinics.net/lists/linux-usb/msg147987.html

ED6E0F17, 2016-11-09:
Closed pending further testing - I assume it does not occur with the default config, and was triggered by enabling CONFIG_PM without CONFIG_SUSPEND .

ED6E0F17, 2016-11-09:
This patch suggests that the test for "(struct->lx_state == DWC2_L2) " in gadget.c should have been replaced with "struct->bus_suspended)"

https://github.com/raspberrypi/linux/commit/1fb7f12d5bec1b0a699c85d724bd5184a01fa2dd

popcornmix, 2016-11-09:
Are you saying there is a bug in the upstream dwc2 driver?

ED6E0F17, 2016-11-09:
There are many bugs in the upstream DWC2, but this seems to fix my current problem:

hardware/bsp/rpi/kernel$ git diff
diff --git a/drivers/usb/dwc2/gadget.c b/drivers/usb/dwc2/gadget.c
index af46adf..d93738a 100644
--- a/drivers/usb/dwc2/gadget.c
+++ b/drivers/usb/dwc2/gadget.c
@@ -848,7 +848,7 @@ static int dwc2_hsotg_ep_queue(struct usb_ep *ep, struct usb_request *req,
                req->zero, req->short_not_ok);
 
        /* Prevent new request submission when controller is suspended */
-       if (hs->lx_state == DWC2_L2) {
+       if (hs->bus_suspended) {
                dev_dbg(hs->dev, "%s: don't submit request while suspended\n",
                                __func__);
                return -EAGAIN;

popcornmix, 2016-11-09:
Submitting upstream to get peer review would be useful. See:
https://www.kernel.org/doc/Documentation/SubmittingPatches

We can cherry-pick any fixes if the response is positive.

ED6E0F17, 2016-11-10:
This fix is clearly wrong, since "bus_suspended" should not compile in gadget-only mode. In dual mode the "lx_state" flag is being left at "DWC2_L2" when resuming as a peripheral - this seems to be deliberate so I am not sure what is correct for the driver

lategoodbye, 2016-11-10:
@ED6E0F17 Are you able to bisect this issue?
Could you also provide the relevant dwc2 kernel config options to reproduce the issue?

ED6E0F17, 2016-11-11:
I will be able to bisect the issue now that I have a better understanding of the code. My first attempt failed when I hit the same issue after "reverting" to 4.7 latest - but that did narrow down the changes substantially.

For reference, the Kernel config I use for Pi Zero is at "https://github.com/ED6E0F17/brillo_device_raspberrypi/blob/master/liz/kernel.config". It is unfortunate that I did not test any kernels for two months, but the only relevant config change I made prior to the failure was adding "CONFIG_PM", which is now required for USB support. CONFIG_PM has been part of downstream defconfig since April.

lategoodbye, 2016-11-11:
Just two question: did you tried to reproduce this issue with mainline kernel? Do you use the downstream bcm2835 devicetree?

ED6E0F17, 2016-11-12:
With the downstream devicetree. I have a B+ running stock kernel and DWC2 in host mode with no issues.

I have not tested Dual Mode with Upstream or the Downstream defconfig but I have run out of other options. Setting "CONFIG_PM" was enough to break my previous 4.6 config. Reverting to 4.4.30 and 20, 40 or 50 USB patches before that did not get back to a working build. Adding Suspend and Hibernate options did not help.

ED6E0F17, 2016-11-14:
Booting with an OTG cable before switching to CDC seems so fix the problem, so I will close this as an upstream issue that may not be present when DWC2 is built as a module.

closed 2016-11-14 by ED6E0F17

ED6E0F17, 2016-11-28:
Setting "usb_autosuspend_delay=-1" in cmdline.txt seems to bypass the issue.

https://git.linaro.org/people/john.stultz/android-dev.git/commit/?h=dev/hikey-4.8&id=6af78ff4760c90f5b5abca6fe77fef8294a231db

lategoodbye, 2016-11-28:
@ED6E0F17 Last week i tried to reproduce your problem with current mainline usb-next. Unfortunately i found issues but not yours.

Is it correct that dr_mode for dwc2 node in DT is still "otg" and not "peripheral"?

Everything would be easier if there was otg and gadget mode support for the Raspberry Pi upstream.

ED6E0F17, 2016-12-01:
@lategoodbye  I should be able to switch to Upstream soon, but there are still some sdcard patches from Anholt that have not been merged. It would be helpful if you could remove `#include "bcm283x-rpi-usb-host.dtsi" ` from the upstream bcm2835-rpi-zero.dtb . Downstream defaults to dr_mode=otg because you are not expected to use dwc2 for pure host mode.

Can you confirm that you use U-boot for upstream work?

lategoodbye, 2016-12-01:
Since there is currently no support for otg or peripheral mode for bcm2835 upstream i decided to switch to host mode via DT. If this fixed i would be happy to switch back to otg mode. I was in contact with John Youn about this issue ( http://marc.info/?l=linux-usb&m=148037984315368&w=2 ). Btw i plan to make a 4.10 test branch with peripheral mode support.

Yes, i'm using U-Boot for upstream work.

ED6E0F17, 2017-01-11:
I have not managed to produce a working upstream image that could replicate this issue, but it is resolved with the pending "dwc2: fixes for 4.10-rc4": "Avoid suspending if we're in gadget mode" cherry-picked into my 4.9 downstream build.

ED6E0F17, 2017-01-26:
Fix merged upstream from 4.9.6   /  4.10-rc5 

Since this issue is not present in Raspbian kernels there should be no benefit from backporting to 4.4

TheDiveO, 2017-04-04:
Tested current stable Raspbian from 2017/03/02 on three RPi0W's: all three(!) show the dmesg error:

`dwc2 20980000.usb: dwc2_hsotg_enqueue_setup: failed queue (-11)`

One RPi0W works, though. Two RPi0W's fail: cannot be pinged and cannot ping.

TheDiveO, 2017-04-04:
Just installed latest Raspbian updates, and the previously failing Pi0W's **now work correctly** in USB Ethernet gadget mode.