raspberrypi/linux #1723: No rule to make target arch/arm/boot/dts/overlays/pisound.dtbo

sfantree, 2016-11-17:
i have cross-complined rpi-4.9 kernel but something wrong when i "make dtbs"

```
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- dtbs
```
it shows :
```
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
 
   ......

make[2]: *** No rule to make target `arch/arm/boot/dts/overlays/pisound.dtbo', needed by `__build'.  Stop.
make[1]: *** [arch/arm/boot/dts/overlays] Error 2
make[1]: *** Waiting for unfinished jobs....
```

pelwell, 2016-11-17:
Thanks - that's shown up two issues:

1) The pisound module hadn't been incorporated into the rpi-4.9.y branch, but cherry-picking the commit that revamped the overlay Makefile from rpi-4.8.y brought in the make rule by accident. The pisound driver has now been cherry-picked into rpi-4.9.y.

2) The DTS and overlay clean rules are failing to delete the .dtbos, which is why we wouldn't have spotted the build error. I'll tackle this later.

pelwell, 2016-11-18:
`make clean` should now remove the .dtbos as well.

Please close this issue if you are satisfied.

closed 2016-11-18 by sfantree